o3de/ROSConDemo #193: Building docker image requires large volume

michalpelka, 2022-11-09:
I was trying to build the ROSConDemo image according to https://github.com/o3de/ROSConDemo/tree/main/docker .
Docker compose consumes all available disk space (about ~ 400 GB) and crashes.
```
docker build -t roscon_demo -f ROSConDemo/docker/Dockerfile .
sending build context to Docker daemon  113.3GB
```

j-rivero, 2022-11-09:
I was able to build it during the merge of #117. I reviewed the efficiency of the image space using [dive](https://github.com/wagoodman/dive) and the total image size here was 44Gb with a reported efficiency of 99%.

[image]

Let me try to build it again without using cache to see if something has change.

j-rivero, 2022-11-09:
I think that I know what's happening: note that the build context is referring to the info sent to the docker daemon and this info is all the directories and files inside the directory from which you are running the `docker build` command, somewhere in your home or code directory probably. Let's change the README to avoid this.

closed 2022-11-10 by adamdbrw