o3de/ROSConDemo #197: ROSConDemo Docker readme.md has full path link from development to main readme.md in a different fork

smurly, 2022-11-11:
https://github.com/o3de/ROSConDemo/blob/development/docker/README.md ends with a full path link to the next step being the demo readme. the link switches branches and fork to main when it should use a relative link to remain in the branch you are starting your docker from.

actual:
`Continue with the instruction in the [main README file](https://github.com/aws-lumberyard/ROSConDemo/blob/main/README.md).`
this link is full path to a different fork's main branch

expected:
it should be a relative path up one directory to README.md if possible so that instruction continue in the docker's branch.

j-rivero, 2022-11-17:
Thanks for pointing that out. I've made a comment to pull #198 to fix this. Let me know if you think that we need to fix it in a separate PR while the PR gets merged.

michalpelka, 2023-01-09:
Both the main and development branch points to  correct o3de repository. Thank you @smurly for raising and @j-rivero for resolving.

smurly, 2023-01-09:
development branch needs to cherry pick the change to relative path as development still links to main
https://raw.githubusercontent.com/o3de/ROSConDemo/development/docker/README.md
`Continue with the instruction in the
[main README file](https://github.com/aws-lumberyard/ROSConDemo/blob/main/README.md).`

michalpelka, 2023-01-10:
Sorry for closing, here is PR #220. 
I also found some more occurrences of situations that you found.
Thanks for re-opening!

closed 2023-01-10 by michalpelka