o3de/ROSConDemo #203: Assigning new 2D Goal Pose to the Kraken while it is apple_gathering causes the Editor to crash when it finishes

LB-ArturZieba, 2022-12-08:
**Describe the bug**
If the Kraken has movement queued in RViz while it is apple_gathering, the Editor will crash. Please note that this can be inconsistent, but provided steps were the most consistent in getting this issue. The following assert was found in Editor.log files of machines where this issue was reproduced:

Assert information:
File:
```
/data/workspace/o3de/Code/Framework/AzCore/./AzCore/std/containers/unordered_map.h
```
Condition:
```
(139916683091136) 'const AZStd::unordered_map::mapped_type &AZStd::unordered_map<AZStd::pair<AppleKraken::EffectorState, AppleKraken::EffectorState>, AZStd::function<void ()>, AppleKraken::TransitionHash>::at(const AZStd::unordered_map::key_type &) const [Key = AZStd::pair<AppleKraken::EffectorState, AppleKraken::EffectorState>, MappedType = AZStd::function<void ()>, Hasher = AppleKraken::TransitionHash, EqualKey = AZStd::equal_to<AZStd::pair<AppleKraken::EffectorState, AppleKraken::EffectorState>>, Allocator = AZStd::allocator]'
Element with key is not present
```
Line: 225

For more information, please refer to the **Video & logs** section.

**Prerequisites**
* Editor launched with Main level opened and in Game Mode. 
* RViz running in the background.
* Terminal ready to run the apple_gather command.

**Steps to reproduce**
1. Move the Kraken using 2D Goal Pose in RViz near a tree.
2. Execute the apple_gather command.
3. While the Kraken is apple_gathering, in RViz set another path using 2D Goal Pose.
4. Wait for the apple_gather to finish.

**Expected behavior**
Editor remains stable.

**Actual behavior**
Editor crashes.

**Video & logs**

https://user-images.githubusercontent.com/86952082/206434847-b97f0c79-4927-4755-9782-22314484f030.mp4

[GPC-602Editor.log](https://github.com/o3de/ROSConDemo/files/10184853/GPC-602Editor.log)
[GPC-610Editor.log](https://github.com/o3de/ROSConDemo/files/10184855/GPC-610Editor.log)

**Found in Branch**
main ([eb54128](https://github.com/o3de/ROSConDemo/commit/eb541287f59a3bd345d8149688325eef43d8cdfd))

**Desktop**
 - Device: PC
 - OS: Linux
 - Version: Ubuntu 20.04.3 LTS
 - CPU AMD Ryzen 5 3600
 - GPU NVIDIA GeForce RTX 2060 SUPER
 - Memory 16GB

michalpelka, 2022-12-15:
I belive that is fixed in the development branch, it is missing the state transition that we added. I cannot recreate it with the development branch.

lumberyard-employee-dm, 2023-01-10:
It looks like the code is using unordered_map::at to return a reference to AZStd::function stored within the [m_allowedTransactionMap](https://github.com/o3de/ROSConDemo/blob/15c8d7c6e989eab8968a4e979c3ff87c745489a7/Project/Gem/Source/ApplePicker/KrakenEffectorComponent.cpp#L504).

It would be safer to either use the `[]` operator, which would add an empty function entry to the map, which will avoid the crash.

michalpelka, 2023-01-11:
This issue was already fixed here : https://github.com/o3de/ROSConDemo/commit/2b46c23366fade2c313b80cbdec36189ae849ea4#diff-80299cd1c98b9872df79376a3921420468a058fea65f0634a02500d0f3469932R438
with this PR #169.

closed 2023-01-11 by michalpelka