o3de/ROSConDemo #204: Errors appear after loading playground level due to missing apple_kraken_5.prefab

LB-JakubSkorupka, 2022-12-08:
**Describe the bug**
After loading the **playground** level included in the project, a popup with 4 error messages is being displayed.
The following errors refer to the **apple_kraken_5.prefab** that is assigned in **Spawner** entity (ROS2 Spawner component), however mentioned prefab file is missing.

**Error 1**:
`PrefabLoader::LoadTemplate - Loading nested instance 'Instance_[5387454034805]' in target Template '21' from Prefab file '/data/workspace/ROSConDemo/Project/Levels/playground/playground.prefab' failed.`

**Error 2**
`PrefabLoader::LoadNestedInstance - Error occurred while loading nested Prefab file 'Assets/Importer/apple_kraken_5.prefab' from Prefab file 'Levels/playground/playground.prefab'.`

**Error 3**
`PrefabLoader::LoadTemplate - Failed to load Prefab file from 'Assets/Importer/apple_kraken_5.prefab'.Error message: 'Failed to open '/data/workspace/ROSConDemo/Project/Assets/Importer/apple_kraken_5.prefab'.'`

**Error 4**
`Full source path for 'Assets/Importer/apple_kraken_5.prefab' could not be determined. Using fallback logic.`

**Steps to reproduce**
Steps to reproduce the behavior:
1. Launch the Editor.
2. Load **playground** level.
3. Select **Spawner** entity (note that **apple_kraken_5.prefab** is assigned in ROS2 Spawner component).
4. In ROS2 Spawner component, try to reassign **apple_kraken_5.prefab**.

**Expected behavior**
Level is loaded without errors, **apple_kraken_5.prefab** is present in the project.

**Actual behavior**
Multiple errors are printed after loading **playground** level.

**Video**

https://user-images.githubusercontent.com/86951967/206439101-30cc0734-f8f0-405d-9faf-c3e40220ef95.mp4

**Found in Branch**
Main (eb54128)

**Desktop:**
 - Device: PC
 - OS: Linux
 - Version: Ubuntu 20.04.5 LTS
 - CPU: Intel Core i7-8700K
 - GPU: NVIDIA GeForce RTX 2080 Ti
 - Memory: 16GB

LB-JakubSkorupka, 2022-12-12:
Updated the Found in Branch with O3DE commit at the time of reporting:
ROSConDemo: main ([eb54128](https://github.com/o3de/ROSConDemo/commit/eb541287f59a3bd345d8149688325eef43d8cdfd))
O3DE: Development ([5c57fee](https://github.com/o3de/o3de/commit/5c57fee7ff1ff592beae42e22f7d7dea814e6912))

closed 2023-01-09 by michalpelka

LB-HubertUberman, 2023-01-18:
Verified as fixed on ROSConDemo: Development (https://github.com/o3de/ROSConDemo/commit/e10e714a011a4c9c2326163d7428c474e32a28e9) and O3DE: Development (https://github.com/o3de/o3de/commit/2bd2ade59e81a4289d81c0283c2ea4c40c26fec9).