o3de/ROSConDemo #205: apple_kraken_v2.prefab starts below the Ground in the playground level

LB-ArturZieba, 2022-12-08:
**Describe the bug**
*apple_kraken_v2.prefab* placed in the *playground* level starts at Translation Z -16, causing it to fall to the void when entering the Game Mode. 

**Steps to reproduce**
1. Launch the playground level.
2. Enter the Game Mode.

**Expected behavior**
Kraken prefab starts above the Ground entity.

**Actual behavior**
Kraken falls to the void.

**Video**

https://user-images.githubusercontent.com/86952082/206450618-0b528dc7-7505-47d2-830b-a4b9abf9254f.mp4

**Found in Branch**
main ([eb54128](https://github.com/o3de/ROSConDemo/commit/eb541287f59a3bd345d8149688325eef43d8cdfd))

**Desktop**
 - Device: PC
 - OS: Linux
 - Version: Ubuntu 20.04.3 LTS
 - CPU AMD Ryzen 5 3600
 - GPU NVIDIA GeForce RTX 2060 SUPER
 - Memory 16GB

LB-ArturZieba, 2022-12-12:
Updated the **Found in Branch** with O3DE commit at the time of reporting:
ROSConDemo: main ([eb54128](https://github.com/o3de/ROSConDemo/commit/eb541287f59a3bd345d8149688325eef43d8cdfd))
O3DE: Development ([5c57fee](https://github.com/o3de/o3de/commit/5c57fee7ff1ff592beae42e22f7d7dea814e6912))

pawelbudziszewski, 2022-12-13:
Thank you for the report. 
This issue should be solved by #213

closed 2023-01-09 by michalpelka