o3de/ROSConDemo #206: Inconsistent Editor freezes after entering Game Mode in Main level

LB-JakubSkorupka, 2022-12-08:
**Describe the bug**
The Editor may freeze after entering Game Mode in **Main** level.
The following issues were noticed on a corresponding machines (dxdiags attached)

 - Freezes are very inconsistent as they seem to happen randomly with about 40% reproduction rate no matter what steps were taken before entering the Game Mode on the same machine

[GPC610_hardinfo_report.txt](https://github.com/o3de/ROSConDemo/files/10185950/GPC610_hardinfo_report.txt)

 - On one machine reassigning Kraken prefabs to the Spawner entity before entering the Game Mode prevents freeze from happening 

[GPC-602_hardinfo_report.txt](https://github.com/o3de/ROSConDemo/files/10185963/GPC-602_hardinfo_report.txt)

 - On another machine the freeze is less likely to occur if the Game Mode was first entered on the Playground level, however after around 10-15 attempts, it was possible to enter Game Mode on first try without entering Playground level earlier.

[GPC-460_hardinfo_report.txt](https://github.com/o3de/ROSConDemo/files/10185999/GPC-460_hardinfo_report.txt)

 - On a single machine the freeze did not occur at all.

[DxDiag_GPC-614.txt](https://github.com/o3de/ROSConDemo/files/10185972/DxDiag_GPC-614.txt)

Note that project was setup by following instruction - https://github.com/o3de/ROSConDemo/tree/main/docker

**Steps to reproduce**
Steps to reproduce the behavior:
1. Launch the Editor.
2. Load the **Main** level.
3. Enter Game Mode (Ctrl + G).

**Expected behavior**
It is possible to enter Game Mode on **Main** level without issues. The Editor remains stable.

**Actual behavior**
The Editor may freeze after entering Game Mode on **Main** level.

**Video**

https://user-images.githubusercontent.com/86951967/206465964-d19cac52-a749-4241-bc41-217a9055bff0.mp4

**Found in Branch**
Main (eb54128)

**Desktop:**
 - Device: PC
 - OS: Linux
 - Version: Ubuntu 20.04.4 LTS
 - CPU: Intel Core i7-8700K
 - GPU: NVIDIA GeForce RTX 2080 Ti
 - Memory: 16GB

LB-JakubSkorupka, 2022-12-12:
Updated the Found in Branch with O3DE commit at the time of reporting:
ROSConDemo: main ([eb54128](https://github.com/o3de/ROSConDemo/commit/eb541287f59a3bd345d8149688325eef43d8cdfd))
O3DE: Development ([5c57fee](https://github.com/o3de/o3de/commit/5c57fee7ff1ff592beae42e22f7d7dea814e6912))

forhalle, 2023-01-10:
@smurly was unable to repo this with latest O3DE dev. Can you please try again with latest?

closed 2023-06-29 by adamdbrw