o3de/ROSConDemo #207: Building ROSConDemo project with docker takes an extremely long time on a specific machine

LB-MateuszSzweda, 2022-12-08:
**Describe the bug**
Building **ROSConDemo** project with docker obtained either by following https://github.com/o3de/ROSConDemo/tree/main/docker guideline or directly from https://docs.docker.com/engine/install/ubuntu/ site takes about 10 hours on specific **Ubuntu 20.04** machine with **AMD Ryzen 9 3900X** CPU.
This is an extremely low performance given that similar machines with **AMD Ryzen 5 3600** CPUs were able to finish the building process in about 1-2 hours.

Please note that **AMD Ryzen 9 3900X** machine does not seem to have any performance issues running the project after 10 hour building finishes and is able to build regular o3de engine in about 1-2 hours by following https://www.o3de.org/docs/welcome-guide/setup/setup-from-github/building-linux/ steps.

The swap file of the machine was changed to 16GB and then 15GB with consecutive building attempts to verify the swap size is not the culprit of the issue.

For further information please refer to the building logs, the machine DXdiag file and Ubuntu lshw output.

**Steps to reproduce**

1. Open the terminal and clone the **ROSConDemo** project with the `git clone git@github.com:aws-lumberyard/ROSConDemo.git` command.
2. Change directory to docker directory with the `cd ROSConDemo/docker` command.
3. Install docker with `sudo snap install docker` command.
4. Build the project via docker with `docker build -t roscon_demo_galactic --build-arg ROS_VERSION=galactic --build-arg UBUNTU_VERSION=focal  .` command.

**Expected behavior**
Building via docker finishes in about 1-3 hours.

**Actual behavior**
Building via docker finishes in about 10 hours.

**Found in Branch**
N/A

**Commit ID from [o3de/o3de](https://github.com/o3de/o3de) Repository**
[eb54128](https://github.com/o3de/ROSConDemo/commit/eb541287f59a3bd345d8149688325eef43d8cdfd)

**Desktop/Device (please complete the following information):**
- Device: PC
- OS: Linux
- Version: Ubuntu 20.04.4 LTS
- CPU: AMD Ryzen 9 3900X
- GPU: NVIDIA GeForce RTX 2060 SUPER
- Memory: 16GB

**Additional context**
[DxDiag_GPC-614.txt](https://github.com/o3de/ROSConDemo/files/10186263/DxDiag_GPC-614.txt)
[Docker building ROSConDemo log after 2 hours.txt](https://github.com/o3de/ROSConDemo/files/10186278/Docker.building.ROSConDemo.log.after.2.hours.txt)
[Docker from web building ROSConDemo log after 2 hours.txt](https://github.com/o3de/ROSConDemo/files/10186281/Docker.from.web.building.ROSConDemo.log.after.2.hours.txt)
[Ubuntu lshw output.txt](https://github.com/o3de/ROSConDemo/files/10186283/Ubuntu.lshw.output.txt)

spham-amzn, 2022-12-12:
Can you also provide the following:
1. The hard drive space available
2. Where docker was installed from?

LB-MateuszSzweda, 2022-12-13:
1. The space available on the hard drive was about 539GB.
2. The docker was installed from two sources which each consecutive try.
Firstly directly by `sudo snap install docker     # version 20.10.17` command.
and directly by following steps https://docs.docker.com/engine/install/ubuntu/ on the second try.

so by using:
```
sudo mkdir -p /etc/apt/keyrings
curl -fsSL https://download.docker.com/linux/ubuntu/gpg | sudo gpg --dearmor -o /etc/apt/keyrings/docker.gpg
```

```
echo \
  "deb [arch=$(dpkg --print-architecture) signed-by=/etc/apt/keyrings/docker.gpg] https://download.docker.com/linux/ubuntu \
  $(lsb_release -cs) stable" | sudo tee /etc/apt/sources.list.d/docker.list > /dev/null 
```
  
` sudo apt-get update`
  
`sudo apt-get install docker-ce docker-ce-cli containerd.io docker-compose-plugin`

commands.

michalpelka, 2023-01-10:
I think I have similar behavior on another machine that has only 16 Gb of RAM. It seems that the build process manages resources quite well, until linking some objects - then it whole process exhaust RAM and start using swap.
I have only a 2 Gb swap so the process got killed by itself. Sorry for the photo but the machine was unresponsive.
[image]
 I am trying to modify the build process to use only 4 jobs.

michalpelka, 2023-01-10:
What is your opinion on that @spham-amzn?

forhalle, 2023-01-10:
@spham-amzn will add some advisory information in the readme (if you have x-GB RAM, then set parallel build jobs to be x/2, etc.), as well as update the Docker script to restrict the number of cores with an option to override.

spham-amzn, 2023-01-31:
The CMAKE_JOBS was introduced into the dockerscript and defaults to 8 at the moment.

LB-MateuszSzweda, 2023-02-08:
I can confirm that introduced change did resolve my issue and I was able to build with `docker build -t roscon_demo_galactic --build-arg ROS_VERSION=galactic --build-arg UBUNTU_VERSION=focal  .` command on the same machine in a timeframe of 2 hours instead of 10 hours.
Linking jobs were performed correctly and in adequate timelapse.

With that the issue can probably be closed.

closed 2023-06-29 by adamdbrw

adamsj-ros, 2023-08-04:
> The CMAKE_JOBS was introduced into the dockerscript and defaults to 8 at the moment.

It is my understanding that CMAKE_JOBS only affects the cmake step. The exhaustion comes at a link step. My machine have 16 cores and even with 8 set as the default in CMAKE_JOBS, the machine was using all 16 cores during the link step of `Editor`.