o3de/ROSConDemo #209: Building ROSConDemo image is consistently blocked by non-zero code:137

LB-ArturZieba, 2022-12-09:
**Describe the bug**
Building either ROSConDemo or o3de-demo-project Docker image consistently fails due to non-zero code: 137. We could not pinpoint exact cause of this, as an example here is a situation from yesterday/today (12/8/2022 & 12/9/2022):
* ROSConDemo Docker image build was blocked due to non-zero code:137.
* After wiping docker from the system, reinstalling it and following steps one by one ROSConDemo image was built and almost entire sample workflow was executed without any bigger issues.
* Attempted to setup the o3de-demo-project for a different workflow and the same problem with non-zero code:137 occured again.
* Wiped docker from the system again, set it up again using the same steps and the first image build (`sudo docker build -t o3de_robot_vacuum_simulation:latest .`) went through without issues.
* Proceeded with the `xhost +local:root` and `sudo docker run --rm --network="bridge" --gpus all -e DISPLAY=:1 -v /tmp/.X11-unix:/tmp/.X11-unix -it o3de_robot_vacuum_simulation:latest /data/workspace/LaunchSimulation.bash` commands, which executed seemingly without issue.
* On executing `sudo docker run --rm --network="bridge" --gpus all -e DISPLAY=:1 -v /tmp/.X11-unix:/tmp/.X11-unix -it o3de_robot_vacuum_navstack:latest /data/workspace/LaunchNavStack.bash` returned an error about missing image, so went back in instruction to the step `You can also create a separate docker image that only contains the navigation stack and Rviz2 by supplying the argument IMAGE_TYPE and setting it to 'navstack': sudo docker build --build-arg IMAGE_TYPE=navstack -t o3de_robot_vacuum_navstack:latest .`. It was not executed earlier as it is described as "**You can** also create a separate docker image **that only contains** the navigation stack and Rviz2" so assumed that it is not required after executing earlier image build.
* Executing `sudo docker run --rm --network="bridge" --gpus all -e DISPLAY=:1 -v /tmp/.X11-unix:/tmp/.X11-unix -it o3de_robot_vacuum_navstack:latest /data/workspace/LaunchNavStack.bash` got blocked again by the non-zero code:137.
* At this point different workarounds were tried, but seemingly every image build (ROSConDemo and both of the images in the o3de-demo-project) is blocked even after Docker restarting/clearing cache and multiple different suggested workarounds.

Different workarounds attempted:
* Removing and adding back 16GB swapfile/restarting PC. (It should be noted that on successful builds there was a 16GB swapfile present already.)
* Removing and adding back 32GB swapfile/restarting PC.
* Running build command with different --memory="<size>" arguments (For example --memory="1g", --memory="2g", --memory="5g", memory="15g").
* Running after apt-get clean, autoclean, autoremove commands.
* Running `sudo docker restart $(sudo docker ps -a -q)`.
* Running `sudo systemctl restart docker`.
* Running `sudo docker system prune -a`.
* Clearing /var/cache/ manually.

Please refer to the **Sample log from Terminal on block** section for sample logs from both ROSConDemo and o3de-demo-project projects. Here is a sample occurence of this error:
```
Preparing to unpack .../135-fonts-dejavu-core_2.37-2build1_all.deb ...
Unpacking fonts-dejavu-core (2.37-2build1) ...
Selecting previously unselected package fontconfig-config.
Preparing to unpack .../136-fontconfig-config_2.13.1-4.2ubuntu5_all.deb ...
Unpacking fontconfig-config (2.13.1-4.2ubuntu5) ...
Killed
The command '/bin/sh -c apt-get update && apt-get install -y     bc     bind9-utils     binutils     ca-certificates     clang-12     cmake     file     firewalld     git     git-lfs     jq     kbd     kmod     less     lsb-release     libglu1-mesa-dev     libxcb-xinerama0     libfontconfig1-dev     libcurl4-openssl-dev     libnvidia-gl-470     libssl-dev     libxcb-xkb-dev     libxkbcommon-x11-dev     libxkbcommon-dev     libxcb-xfixes0-dev     libxcb-xinput-dev     libxcb-xinput0     libpcre2-16-0     lsof     net-tools     ninja-build     pciutils     python3-pip     software-properties-common     sudo     tar     unzip     vim     wget     xz-utils     && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 137
```

This issue effectively blocks any further progress and should be considered a blocker.

**Steps to reproduce**
1. Follow steps on either https://github.com/o3de/ROSConDemo/blob/main/docker/README.md or https://github.com/RobotecAI/o3de-demo-project/blob/main/Docker/README.md up to and including Docker image build.

**Expected behavior**
Docker image gets built reliably.

**Actual behavior**
Docker image build gets blocked due to non-zero code: 137 consistently.

**Sample log from Terminal on block**
[GPC610ROSConDemoBuildBlock.txt](https://github.com/o3de/ROSConDemo/files/10193783/GPC610ROSConDemoBuildBlock.txt)
[GPC610o3de-demo-projectBuildBlock.txt](https://github.com/o3de/ROSConDemo/files/10193784/GPC610o3de-demo-projectBuildBlock.txt)

**Found in Branch**
main ([eb54128](https://github.com/o3de/ROSConDemo/commit/eb541287f59a3bd345d8149688325eef43d8cdfd))

**Desktop**
 - Device: PC
 - OS: Linux
 - Version: Ubuntu 20.04.3 LTS
 - CPU AMD Ryzen 5 3600
 - GPU NVIDIA GeForce RTX 2060 SUPER
 - Memory 16GB (additionally tested with 16GB and 32GB Swapfile)

LB-ArturZieba, 2022-12-12:
Updated the **Found in Branch** with O3DE and o3de-demo-project commits at the time of reporting:
ROSConDemo: main ([eb54128](https://github.com/o3de/ROSConDemo/commit/eb541287f59a3bd345d8149688325eef43d8cdfd))
o3de-demo-project: main ([a1d8921](a1d892103c20aed254d677fc46917ce189254271))
O3DE: Development ([43ecf18](43ecf18cd1297703fb540e0df5f8d880b68e5541))

LB-ArturZieba, 2023-01-10:
I have rechecked this today on Development (aba3ba9) and the issue still occurs even after removing all docker images and making sure that there is enough disk space (~380GB) and swapfile was set to 32GB. Here are the details of what I have verified:

* Used `sudo docker images` and `sudo docker image rm --force <IMAGE ID>` commands to remove all existing docker images one by one.
* Used `swapon -s` command to verify swapfile size.
* Rechecked the above and checked available disk space - no images were found by `sudo docker image`, swapfile size was 32.0GB, available disk space was 388.8 GB.
* Cloned the repository to /home/gpc-610/sor/ folder - `git clone https://github.com/o3de/ROSConDemo.git --branch development` (adjusted repository compared to https://github.com/o3de/ROSConDemo/blob/development/docker/README.md where it is `git clone git@github.com:aws-lumberyard/ROSConDemo.git`).
* Executed `sudo docker build -t roscon_demo -f Dockerfile .` command in the /home/gpc-610/sor/ROSConDemo/docker/ folder - Killed non-zero code:137 - can be seen on the attached DockerBuild1.mp4 and DockerBuild1Log.txt files.
* Executed `sudo docker build -t roscon_demo_galactic --build-arg ROS_VERSION=galactic --build-arg UBUNTU_VERSION=focal .` command in the /home/gpc-610/sor/ROSConDemo/docker/ folder - Killed non-zero code:137 - can be seen on the attached DockerBuild2.mp4 and DockerBuild2Log.txt files. (Command from https://github.com/o3de/ROSConDemo/blob/main/docker/README.md)
* Removed Dockerfile from /home/gpc-610/sor/ROSConDemo/docker/ folder and executed commands `wget https://raw.githubusercontent.com/o3de/ROSConDemo/main/docker/Dockerfile` and `docker build -t roscon_demo_galactic --build-arg ROS_VERSION=galactic --build-arg UBUNTU_VERSION=focal .` from https://github.com/o3de/ROSConDemo/blob/main/docker/README.md - Killed non-zero code:137 - can be seen on the attached DockerBuild3.mp4 and DockerBuild3Log.txt files.
* Removed the /home/gpc-610/sor/ folder.
* Created a new /home/gpc-610/asor/ folder and opened Terminal in it.
* Removed all docker images again.
* Cloned the repository to /home/gpc-610/asor/ folder - `git clone https://github.com/o3de/ROSConDemo.git --branch development`.
* Executed `sudo docker build -t roscon_demo -f Dockerfile .` command in the /home/gpc-610/sor/ROSConDemo/docker/ folder - Killed non-zero code:137 - can be seen on the attached DockerBuild4.mp4 and DockerBuild4Log.txt files.

**Videos, log files and a screenshot of initial setup**:
[image]

https://user-images.githubusercontent.com/86952082/211522756-908326f5-9d76-4f90-8f92-f6ede19a36fa.mp4

[DockerBuild1Log.txt](https://github.com/o3de/ROSConDemo/files/10381426/DockerBuild1Log.txt)

https://user-images.githubusercontent.com/86952082/211522743-de731e72-e2a0-467f-89a5-2978377ee9ff.mp4

[DockerBuild2Log.txt](https://github.com/o3de/ROSConDemo/files/10381424/DockerBuild2Log.txt)

https://user-images.githubusercontent.com/86952082/211522720-0e06b9d3-7644-4c3b-9d5f-e10ee51dca52.mp4

[DockerBuild3Log.txt](https://github.com/o3de/ROSConDemo/files/10381423/DockerBuild3Log.txt)

https://user-images.githubusercontent.com/86952082/211522769-246e6348-e779-4dc4-9406-168befaa470c.mp4

[DockerBuild4Log.txt](https://github.com/o3de/ROSConDemo/files/10381422/DockerBuild4Log.txt)

LB-ArturZieba, 2023-01-10:
@michalpelka Running `rocker --nvidia --x11 osrf/ros:melodic-desktop-full gazebo` launched Gazebo for me, attaching screenshot and a log below:
[image]

[GazeboLog.txt](https://github.com/o3de/ROSConDemo/files/10383358/GazeboLog.txt)

As for `nvidia-smi`, here is the log from that:
[GPC-610nvidia-smi.txt](https://github.com/o3de/ROSConDemo/files/10383360/GPC-610nvidia-smi.txt)

michalpelka, 2023-01-10:
Thank you for checking @LB-ArturZieba .
Sorry to bother you again, but I have another check to run. I can see that list of packages installed by the docker build is redundant.
I prepared dockerfile that installs minimal packages, just before usage. 
My main suspect is nvidia vulkan libraries (libnvidia-gl-470).
Here is my branch : https://github.com/michalpelka/ROSConDemo/blob/mp/adjust_docker_packages/docker/Dockerfile .
Do you have some setup procedure on your test machine - it would be beneficial to recreate it on our site!
Thanks!

forhalle, 2023-01-10:
@michalpelka - This appears to be a Docker setup issue, and not an issue with O3DE code itself, although we need to ensure our setup instructions are robust.

@LB-ArturZieba - To be successful, you can only have one Docker install on your machine (and it should be the NVIDIA Docker). To confirm this is the case, please uninstall all Docker packages (not images, although you can do that to if you like), and perform a fresh install of the [NVIDIA Docker]([url](https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/install-guide.html)).

LB-ArturZieba, 2023-02-13:
This is no longer reproducible on our side (on the same machine on which it was originally reported) after upgrading to Ubuntu 22.04.1 LTS. Closing the issue.

closed 2023-02-13 by LB-ArturZieba