o3de/ROSConDemo #210: Editor crashes when opening a level

LB-KonradJuszczak, 2022-12-09:
**Describe the bug**
While opening newly created or already existing level in O3DE Editor, the engine crashes with a prompt in terminal: Segmentation fault (core dumped).

Note that project was setup by following the instruction: https://github.com/o3de/ROSConDemo/tree/main/docker

Please refer to the Editor.log provided as an attached file. Also please look at the hardinfo report from both PCs on which the problem occurs.
- [Hardinfo.zip](https://github.com/o3de/ROSConDemo/files/10195496/Hardinfo.zip)
- [Editor.log](https://github.com/o3de/ROSConDemo/files/10195456/Editor.log)

**Steps to reproduce**
1. Launch the Editor.
2. Attempt to open any level.

**Expected behavior**
It is possible to open any level without problems.

**Actual behavior**
Editor crashes when opening a level.

**Found in Branch**
main (eb54128)

**Desktop**
- Device: PC
- OS: Linux
- Version: Ubuntu 20.04.5 LTS
- CPU: AMD Ryzen 5 3600
- GPU: NVIDIA GeForce RTX 2060 SUPERS
- Memory: 16 GB

LB-KonradJuszczak, 2022-12-12:
Updated the Found in Branch with O3DE commit at the time of reporting:
ROSConDemo: main ([eb54128](https://github.com/o3de/ROSConDemo/commit/eb541287f59a3bd345d8149688325eef43d8cdfd))
O3DE: Development ([43ecf18](https://github.com/o3de/o3de/commit/43ecf18cd1297703fb540e0df5f8d880b68e5541))

lumberyard-employee-dm, 2022-12-12:
This issue might be related to an issue with the Editor crashing before the AssetProcessor processing all the critical assets: https://github.com/o3de/o3de/commit/17213adaaa1febeafafe8e351dcdd4dc7d336ee8

forhalle, 2023-01-09:
@spham-amzn will verify this issue and close if it can't be reproduced. @smurly was unable to reproduce on his machine using a subsequent commit.

hultonha, 2023-01-31:
@smurly Is this something that can be reviewed and closed as it seems to no longer be occurring. The development branch should be used.

adamdbrw, 2023-02-10:
@spham-amzn any luck reproducing this one?

LB-TomaszSerwanski, 2023-02-10:
Just a quick update - we have verified that this is no longer reproducible on our side (on the same machine on which it was originally reported) after upgrading to Ubuntu 22.04. Please note that closing the issue on our side is not possible due to lack of permissions in the ROSConDemo repository.

adamdbrw, 2023-02-10:
Thank you for verifying  @LB-TomaszSerwanski!

closed 2023-02-10 by adamdbrw