o3de/ROSConDemo #212: ROSConDemo Docker readme.md initial build command should specify sudo to avoid errors

smurly, 2022-12-13:
following the readme.md instructions for docker results in an error when sudo is not specified
```
~/git/ROSConDemo/docker$ docker build -t roscon_demo -f Dockerfile .
Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post "http://%2Fvar%2Frun%2Fdocker.sock/v1.24/build?buildargs=%7B%7D&cachefrom=%5B%5D&cgroupparent=&cpuperiod=0&cpuquota=0&cpusetcpus=&cpusetmems=&cpushares=0&dockerfile=Dockerfile&labels=%7B%7D&memory=0&memswap=0&networkmode=default&rm=1&shmsize=0&t=roscon_demo&target=&ulimits=null&version=1": dial unix /var/run/docker.sock: connect: permission denied
```

specifying `sudo docker build -t roscon_demo -f Dockerfile .` will proceed without permission denied

michalpelka, 2022-12-15:
I would not recommend using docker with root privileges, since I can see no benefit.
I think there is a problem with your environment, please refer to [Post Instalation Steps for Docker](https://docs.docker.com/engine/install/linux-postinstall/) .
I think the User on your machine is not in Docker group. You can verify that with `groups` :
```
michal@michal-AORUS-15P-XD:~$ groups
michal adm cdrom sudo dip plugdev lpadmin lxd sambashare docker
```

smurly, 2023-01-19:
we can link to the instructions for setting up docker permissions to possibly avoid sudo

spham-amzn, 2023-01-19:
Yes, that was done for the RobotVacuum demo, we can do it for here
https://github.com/o3de/RobotVacuumSample/commit/e206ba86d01863f5985b263afcc0b66078e239eb

closed 2023-01-26 by spham-amzn