o3de/ROSConDemo #214: ROSConDem Docker o3de_kraken_nav doesn't seem to function even if build in the docker

smurly, 2022-12-13:
starting from https://github.com/o3de/ROSConDemo/tree/development/docker using rocker, I'm able to get the editor loaded with the main level and successfully spawn a robot, but navstack fails to launch
```
root@9c3dd878bb1d:/data/workspace/o3de_kraken_ws/build/o3de_kraken_nav# ros2 service call /spawn_entity gazebo_msgs/srv/SpawnEntity '{name: 'apple_kraken_rusty', xml: 'line1'}'
requester: making request: gazebo_msgs.srv.SpawnEntity_Request(name='apple_kraken_rusty', xml='line1', robot_namespace='', initial_pose=geometry_msgs.msg.Pose(position=geometry_msgs.msg.Point(x=0.0, y=0.0, z=0.0), orientation=geometry_msgs.msg.Quaternion(x=0.0, y=0.0, z=0.0, w=1.0)), reference_frame='')

response:
gazebo_msgs.srv.SpawnEntity_Response(success=True, status_message='')

root@9c3dd878bb1d:/data/workspace/o3de_kraken_ws/build/o3de_kraken_nav# ros2 launch o3de_kraken_nav navigation_multi.launch.py namespace:=apple_kraken_rusty_1 rviz:=True
Package 'o3de_kraken_nav' not found: "package 'o3de_kraken_nav' not found, searching: ['/opt/ros/humble']"
```

michalpelka, 2022-12-15:
Thanks,  for checking it out. I think that there is one important step missing in https://github.com/o3de/ROSConDemo/tree/development/docker .
Please check if the problem persists if you correctly source ROS workspace overlay:
```bash
root@4c345a5be212:/data/workspace# source ./o3de_kraken_ws/install/setup.bash```
If the problem is caused by that I will provide PR to documentation.

smurly, 2022-12-15:
this fixed the issue. should the docker image itself be fixed or documentation?

forhalle, 2023-01-09:
@michalpelka  - We should update the documentation to instruct users to open multiple Rocker terminals (one for each robot, one for the simulation, and one to send commands), and run the source command (mentioned above) for each terminal opened. I have reassigned the issue to @spham-amzn  - He will take care of it. Thanks!

closed 2023-01-26 by spham-amzn