o3de/ROSConDemo #218: ROSConDemo Docker build fails ('ParamaterNames' problem)

pawelbudziszewski, 2023-01-10:
```
In file included from /data/workspace/ROSConDemo/Project/build/linux/External/o3de-ros2-gem-2c79e140/Code/CMakeFiles/ROS2.Editor.Static.dir/Unity/unity_1_cxx.cxx:7:
/data/workspace/o3de-ros2-gem/Code/Source/RobotImporter/URDF/JointsMaker.cpp:63:55: error: no member named 'ParamaterNames' in namespace 'PhysX::JointsComponentModeCommon'; did you mean 'ParameterNames'?
                    PhysX::JointsComponentModeCommon::ParamaterNames::Rotation,
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
                                                      ParameterNames
/data/workspace/o3de/Gems/PhysX/Code/./Editor/Source/ComponentModes/Joints/JointsComponentModeCommon.h:47:12: note: 'ParameterNames' declared here
    struct ParameterNames
           ^
In file included from /data/workspace/ROSConDemo/Project/build/linux/External/o3de-ros2-gem-2c79e140/Code/CMakeFiles/ROS2.Editor.Static.dir/Unity/unity_1_cxx.cxx:7:
/data/workspace/o3de-ros2-gem/Code/Source/RobotImporter/URDF/JointsMaker.cpp:69:55: error: no member named 'ParamaterNames' in namespace 'PhysX::JointsComponentModeCommon'; did you mean 'ParameterNames'?
                    PhysX::JointsComponentModeCommon::ParamaterNames::LinearLimits,
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
                                                      ParameterNames
/data/workspace/o3de/Gems/PhysX/Code/./Editor/Source/ComponentModes/Joints/JointsComponentModeCommon.h:47:12: note: 'ParameterNames' declared here
    struct ParameterNames
           ^
In file included from /data/workspace/ROSConDemo/Project/build/linux/External/o3de-ros2-gem-2c79e140/Code/CMakeFiles/ROS2.Editor.Static.dir/Unity/unity_1_cxx.cxx:7:
/data/workspace/o3de-ros2-gem/Code/Source/RobotImporter/URDF/JointsMaker.cpp:83:55: error: no member named 'ParamaterNames' in namespace 'PhysX::JointsComponentModeCommon'; did you mean 'ParameterNames'?
                    PhysX::JointsComponentModeCommon::ParamaterNames::Rotation,
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
                                                      ParameterNames
/data/workspace/o3de/Gems/PhysX/Code/./Editor/Source/ComponentModes/Joints/JointsComponentModeCommon.h:47:12: note: 'ParameterNames' declared here
    struct ParameterNames
           ^
In file included from /data/workspace/ROSConDemo/Project/build/linux/External/o3de-ros2-gem-2c79e140/Code/CMakeFiles/ROS2.Editor.Static.dir/Unity/unity_1_cxx.cxx:7:
/data/workspace/o3de-ros2-gem/Code/Source/RobotImporter/URDF/JointsMaker.cpp:89:55: error: no member named 'ParamaterNames' in namespace 'PhysX::JointsComponentModeCommon'; did you mean 'ParameterNames'?
                    PhysX::JointsComponentModeCommon::ParamaterNames::TwistLimits,
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
                                                      ParameterNames
/data/workspace/o3de/Gems/PhysX/Code/./Editor/Source/ComponentModes/Joints/JointsComponentModeCommon.h:47:12: note: 'ParameterNames' declared here
    struct ParameterNames
           ^
In file included from /data/workspace/ROSConDemo/Project/build/linux/External/o3de-ros2-gem-2c79e140/Code/CMakeFiles/ROS2.Editor.Static.dir/Unity/unity_1_cxx.cxx:7:
/data/workspace/o3de-ros2-gem/Code/Source/RobotImporter/URDF/JointsMaker.cpp:95:55: error: no member named 'ParamaterNames' in namespace 'PhysX::JointsComponentModeCommon'; did you mean 'ParameterNames'?
                    PhysX::JointsComponentModeCommon::ParamaterNames::EnableLimits,
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
                                                      ParameterNames
/data/workspace/o3de/Gems/PhysX/Code/./Editor/Source/ComponentModes/Joints/JointsComponentModeCommon.h:47:12: note: 'ParameterNames' declared here
    struct ParameterNames
           ^
In file included from /data/workspace/ROSConDemo/Project/build/linux/External/o3de-ros2-gem-2c79e140/Code/CMakeFiles/ROS2.Editor.Static.dir/Unity/unity_1_cxx.cxx:7:
/data/workspace/o3de-ros2-gem/Code/Source/RobotImporter/URDF/JointsMaker.cpp:118:55: error: no member named 'ParamaterNames' in namespace 'PhysX::JointsComponentModeCommon'; did you mean 'ParameterNames'?
                    PhysX::JointsComponentModeCommon::ParamaterNames::Rotation,
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
                                                      ParameterNames
/data/workspace/o3de/Gems/PhysX/Code/./Editor/Source/ComponentModes/Joints/JointsComponentModeCommon.h:47:12: note: 'ParameterNames' declared here
    struct ParameterNames
           ^
In file included from /data/workspace/ROSConDemo/Project/build/linux/External/o3de-ros2-gem-2c79e140/Code/CMakeFiles/ROS2.Editor.Static.dir/Unity/unity_1_cxx.cxx:7:
/data/workspace/o3de-ros2-gem/Code/Source/RobotImporter/URDF/JointsMaker.cpp:124:55: error: no member named 'ParamaterNames' in namespace 'PhysX::JointsComponentModeCommon'; did you mean 'ParameterNames'?
                    PhysX::JointsComponentModeCommon::ParamaterNames::TwistLimits,
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
                                                      ParameterNames
/data/workspace/o3de/Gems/PhysX/Code/./Editor/Source/ComponentModes/Joints/JointsComponentModeCommon.h:47:12: note: 'ParameterNames' declared here
    struct ParameterNames
           ^
In file included from /data/workspace/ROSConDemo/Project/build/linux/External/o3de-ros2-gem-2c79e140/Code/CMakeFiles/ROS2.Editor.Static.dir/Unity/unity_1_cxx.cxx:7:
/data/workspace/o3de-ros2-gem/Code/Source/RobotImporter/URDF/JointsMaker.cpp:139:47: error: no member named 'ParamaterNames' in namespace 'PhysX::JointsComponentModeCommon'; did you mean 'ParameterNames'?
            PhysX::JointsComponentModeCommon::ParamaterNames::LeadEntity,
            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
                                              ParameterNames
/data/workspace/o3de/Gems/PhysX/Code/./Editor/Source/ComponentModes/Joints/JointsComponentModeCommon.h:47:12: note: 'ParameterNames' declared here
    struct ParameterNames
           ^
8 errors generated.
```

michalpelka, 2023-01-10:
It is being fixed with #217..

michalpelka, 2023-01-10:
It was fixed by switching the ROS2 gem from o3de-extras. 
It builds and runs correctly now against the current development of O3DE.