o3de/ROSConDemo #225: docker image fails to communicate between rocker terminals can't spawn kraken or use rviz

smurly, 2023-02-01:
using ROSConDemo commit 779e3d4e5af8de7f5715875eb47d52a6a0f14f49
I'm unable to spawn a kraken or get visuals in rviz with multiple rocker terminals against the docker image as instructed in the readme.
G4dn.2Xlarge running ubuntu 22.04
editor and game launcher both work and display the orchard.

I build the image as instructed in the readme
`docker build --build-arg O3DE_BRANCH=199205f --build-arg O3DE_EXTRAS_BRANCH=cbd3cd5 -t roscon_demo -f Dockerfile .`
I also tried again with an old commit set from January 12th
`docker build --build-arg O3DE_BRANCH=d9efce5 --build-arg O3DE_EXTRAS_BRANCH=810279a -t roscon_demo -f Dockerfile .`
3 rocker terminals were open using the readme command `rocker --x11 --nvidia roscon_demo` I tried all of the following additional:
`source /opt/ros/humble/setup.bash`
from the o3de_navigation folder `source ./kraken_nav/install/setup.bash` and `export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp`
both editor and game launcher were tried in one terminal
in another terminal rviz was launched
`ros2 launch o3de_kraken_nav navigation_multi.launch.py namespace:=apple_kraken_rusty_1 rviz:=True`
then in another terminal the command was issued to spawn a robot into game mode
```
root@3e26aa6e4bb4:/data/workspace/ROSConDemo# ros2 service call /spawn_entity gazebo_msgs/srv/SpawnEntity '{name: 'apple_kraken_rusty', xml: 'line1'}'
requester: making request: gazebo_msgs.srv.SpawnEntity_Request(name='apple_kraken_rusty', xml='line1', robot_namespace='', initial_pose=geometry_msgs.msg.Pose(position=geometry_msgs.msg.Point(x=0.0, y=0.0, z=0.0), orientation=geometry_msgs.msg.Quaternion(x=0.0, y=0.0, z=0.0, w=1.0)), reference_frame='')

response:
gazebo_msgs.srv.SpawnEntity_Response(success=True, status_message='')
```
despite success=True no robot spawned and rviz never updated a navigation view.

hultonha, 2023-02-03:
Hey @spham-amzn, just checking if this is on your radar. I think @smurly mentioned this is a recent regression, is this something you could take a look at or let me know about when you have a moment. Thanks!

spham-amzn, 2023-02-03:
I was working with @smurly on this, and it may be an environment setup issue, not an issue with the code itself. I was able to repro on one machine but not another.

hultonha, 2023-02-06:
I see, thanks for the info. If it's hard to reproduce it might be worth closing and seeing if it happens again. We could ask to see if @michalpelka or @pawelbudziszewski have run into anything similar too

smurly, 2023-02-06:
I setup a physical machine but the resulting docker was unable to load the main level without hanging. it seems to get stuck in a "O3DE Editor" is not responding loop. The viewport shows nothing but a black ground plane and sky; no content meshes are resolved.
physical ubuntu 22.04 has 32GB memory available so I'm not sure why it failed

hultonha, 2023-02-07:
Oh this could be related to the recent Static Rigid Body change (looking at the log above) @moraaar Could you point @smurly at the instructions for how to fix this please (update the level prefab) which might resolve the problem. Thanks!

moraaar, 2023-02-07:
> I setup a physical machine but the resulting docker was unable to load the main level without hanging. it seems to get stuck in a "O3DE Editor" is not responding loop. The viewport shows nothing but a black ground plane and sky; no content meshes are resolved.
> 
> physical ubuntu 22.04 has 32GB memory available so I'm not sure why it failed I rebuilt the docker image and saw similar isues with unresponsive editor but noticed an editor error window this time [image]

@smurly please sync to the latest from *ROSConDemo*, *o3de-extras* (for ROS2 gems) and the *o3de* engine you are using it with. A new Static Rigid Body component was added to *o3de* and the prefabs of *ROSConDemo* and *ROS2* gem were updated to use them. From the error messages in the picture it seems like you got the latest from *ROSConDemo* but not from *o3de* and/or *o3de-extras*.

Getting the latest in all the repos involved (*ROSConDemo*, *o3de-extras* and *o3de*) should solve the issue from the picture (which seems non related to the original issue in the description of this GHI).

smurly, 2023-02-07:
I'll see about getting the docker readme updated then. we are specifying specific commits for o3de and o3de-extras
`docker build --build-arg O3DE_BRANCH=199205f --build-arg O3DE_EXTRAS_BRANCH=cbd3cd5 -t roscon_demo -f Dockerfile .`

hultonha, 2023-02-08:
@LB-ArturZieba Maybe you could reference the commits you recently used here and had success with? Thanks!

LB-ArturZieba, 2023-02-08:
These are the commits I've confirmed to be working: 

* Repository clone: development (https://github.com/o3de/ROSConDemo/commit/c4594df2abc42b689193a3f9b5cadc3baa6fa963)
* Image build command with o3de/o3de-extras commits: `docker build --build-arg O3DE_BRANCH=b24ae00 --build-arg O3DE_EXTRAS_BRANCH=d18a96f -t roscon_demo -f Dockerfile .`

hultonha, 2023-02-08:
That's awesome thanks @LB-ArturZieba 👍

smurly, 2023-02-13:
I was able to get this working with commits built as `docker build --build-arg O3DE_BRANCH=ff4c065 --build-arg O3DE_EXTRAS_BRANCH=4c13bbf -t roscon_demo -f Dockerfile .`
the issue appears to have been caused by a prefab system change in ROSConDemo content which required matching o3de and o3de-extras matching changes