o3de/ROSConDemo #229: ROSConDemo docker image is loaded with black ground texture when launched via launcher

LB-HubertUberman, 2023-02-07:
**Describe the bug**
Launching ROSConDemo docker image via launcher results in level being loaded with a black ground texture. Note that after pressing CTRL+R to reset the level issue no longer occurs.

This issue should be considered as a Major.

**Steps to reproduce**
1. Follow steps on https://github.com/o3de/ROSConDemo/blob/development/docker/README.md and build the docker image.
2. Launch the O3DE simulation launcher `/data/workspace/ROSConDemo/Project/build/linux/bin/profile/./ROSConDemo.GameLauncher -LoadLevel=main`

**Expected behavior**
Level is loaded without any graphical issues.

**Actual behavior**
Level is loaded with a black ground texture.

**Video**

https://user-images.githubusercontent.com/96232062/217220391-4cbd34dc-ce29-4dc0-ba98-1068f6441f76.mp4

**Found in Branch**
ROSConDemo: Development (https://github.com/o3de/ROSConDemo/commit/c4594df2abc42b689193a3f9b5cadc3baa6fa963)
O3DE: Development (https://github.com/o3de/o3de/commit/20da779ccc8e5b44c4fd98213298b9295143478c)

**Desktop/Device:**
- Device: PC
- OS: Ubuntu
- Version: 22.04.01
- CPU: AMD Ryzen 5 3600
- GPU: NVIDIA GeForce RTX 2060 SUPER
- Memory: 16GB

adamdbrw, 2023-02-10:
@spham-amzn this does not look like the demo project issue, but rather an O3DE issue (perhaps associated with the Terrain Gem?). Do you have an insight on what could be the problem?

spham-amzn, 2023-02-21:
I have noticed this as well, I will need to troubleshoot the root cause of this.

adamdbrw, 2023-08-31:
@spham-amzn did you manage to troubleshoot the cause?

spham-amzn, 2023-08-31:
I haven't had a chance to look at this in a while. I'll find some time to see the behavior based on the current development (or stabilization)

michalpelka, 2023-12-07:
In recent tests against stabilization branches (2310 and 2310.1) we did not observe this issue, I am closing the issue for now. Thanks!

closed 2023-12-07 by michalpelka