o3de/ROSConDemo #230: Kraken_nav readme.md incorrectly states that Ubuntu 20.04 (Focal) is supported

LB-HubertUberman, 2023-02-08:
## Describe the issue
The [Packages](https://github.com/o3de/ROSConDemo/blob/development/kraken_nav/README.md#packages) subsection in ROS2 prerequisites section of the [Kraken_nav README.md](https://github.com/o3de/ROSConDemo/blob/development/kraken_nav/README.md) file states that Ubuntu 20.04 (Focal) or 22.04 (Jammy) are supported, while only the latter is actually supported by ROSConDemo.

Please note that this issue is considered as a Major.

## Impacted page(s)
[Kraken_nav README.md](https://github.com/o3de/ROSConDemo/blob/development/kraken_nav/README.md)

## Expected experience
The [Kraken_nav README.md](https://github.com/o3de/ROSConDemo/blob/development/kraken_nav/README.md) file states that only Ubuntu 22.04 (Jammy) is supported.

adamdbrw, 2023-02-10:
@LB-HubertUberman to be exact, navigation package is separate from the simulation and can run with any version of ROS which it builds with (either Galactic and Humble) an communicate with the simulation (which runs on Humble) through publish and subscribe. So it should work with Galactic in principle. 

I do not believe it is a major issue since users land on the Demo page first, install the Humble (if not already there) and they are highly unlikely to install the older version as they go through the instructions and land on navigation README.

However, for the sake of clarity and simplicity, I will remove 20.04 from the README.

closed 2023-02-10 by spham-amzn

LB-HubertUberman, 2023-02-13:
Issue verified as fixed on Development (https://github.com/o3de/ROSConDemo/commit/4deca9e39101480af908ea90ec0587f02ac83a6b)

PR #231 (2023-02-10): Remove Galactic from instructions
Includes minor corrections (e.g. ROS2 -> ROS 2).

Fixes #230