o3de/ROSConDemo #235: Dockerscript should not need to clone the repo that it is in

spham-amzn, 2023-02-21:
Currently, the Dockerfile designed for this project will clone from the same repo that it is currently already cloned in. 
```
ARG ROSCON_DEMO_REPO=https://github.com/o3de/ROSConDemo.git
ARG ROSCON_DEMO_BRANCH=development
```
The disadvantage of this is that you cannot make any changes or modifications to test with a Dockerfile unless the changes were merged into the repo. Instead, the Dockerfile should copy the current source into the image instead, but still clone from the other repos

jhanca-robotecai, 2023-06-30:
I faced the same question while implementing [docker environment for Ros2ProjectTemplate](https://github.com/o3de/o3de-extras/pull/357) and I decided to keep it as it is. The docker arguments that you showed allow you to change the repo (you can use your own fork) and the branch. Please note, that you can even use a commit's hash code rather than a branch name:
```
        && git clone $ROSCON_DEMO_REPO \
        && git -C $WORKSPACE/ROSConDemo checkout $ROSCON_DEMO_BRANCH \
```

adamsj-ros, 2023-08-03:
Why not just add `o3de` and `o3de-extras` as submodules and then `COPY .  /data/workspace`? Alternatively, clone only `o3de` and `o3de-extras` still using the `COPY` command in the `Dockerfile`.

spham-amzn, 2024-02-07:
I think we can close this issue. The intent of docker is to provide a consistent environment that is set against known tags if possible. I think this request ultimately defeats that purpose. While the dockerscript itself exists within the same repo that it clones, it should be able to stand alone outside of the repo.

closed 2024-02-07 by spham-amzn