o3de/ROSConDemo #257: assert Vulkan API method failed and error RHI device

adamsj-ros, 2023-08-03:
```
 129   │ SettingsRegistryMergeUtils: Optional config file "/data/workspace/ROSConDemo/Project/Cache/linux/user.cfg" not found.
 130   │ System: Network layer initialized
 131   │ ROS2SystemComponent: Enabling realtime clockSystem: AudioSystem created!
 132   │ AWSCoreConfiguration:
 133   │ ==================================================================
 134   │ AWSCoreConfiguration: Trace::Warning
 135   │  /data/workspace/o3de/Gems/AWSCore/Code/Source/Configuration/AWSCoreConfiguration.cpp(114): 'void AWSCore::AWSCoreConfigurat
       │ ion::ParseSettingsRegistryValues()'
 136   │ AWSCoreConfiguration: Failed to get AllowAWSMetadataCredentials setting, will default to false.
 137   │ AWSCoreConfiguration: ==================================================================
 138   │ RHISystem: Initializing RHI...
 139   │ RHISystem:  Enumerated physical device: llvmpipe (LLVM 15.0.7, 256 bits)
 140   │ RHISystem:  Using physical device: llvmpipe (LLVM 15.0.7, 256 bits)
 141   │ System:
 142   │ ==================================================================
 143   │ System: Trace::Assert
 144   │  /data/workspace/o3de/Gems/Atom/RHI/Vulkan/Code/Source/RHI/Vulkan.h(96): (140400258938048) 'void AZ::Vulkan::AssertSuccess(V
       │ kResult)'
 145   │ System: ASSERT: Vulkan API method failed: Feature not present
 146   │ System: ------------------------------------------------
 147   │ System: AZ::Vulkan::Device::InitInternal(AZ::RHI::PhysicalDevice&) (+0xc5d) [0x7fb10b0e08ed]
 148   │ System: AZ::RHI::Device::Init(int, AZ::RHI::PhysicalDevice&) (+0xe5) [0x7fb0fcb82f55]
 149   │ System: AZ::RHI::RHISystem::InitInternalDevices() (+0x9af) [0x7fb0fcb20a7f]
 150   │ System: AZ::RHI::RHISystem::InitDevice() (+0x2f) [0x7fb0fcb200af]
 151   │ System: AZ::RPI::RPISystem::Initialize(AZ::RPI::RPISystemDescriptor const&) (+0xd5) [0x7fb0fc7caab5]
 152   │ System: AZ::RPI::RPISystemComponent::Activate() (+0x254) [0x7fb0fc4e25c4]
 153   │ System: AZ::ModuleManager::ActivateEntities(AZStd::vector<AZStd::shared_ptr<AZ::ModuleDataImpl>, AZStd::allocator> const&) (
       │ +0xb16) [0x7fb176ab0e76]
 154   │ System: non-virtual thunk to AZ::ModuleManager::OnEntityActivated(AZ::EntityId const&) (+0x94) [0x7fb176ab1804]
 155   │ System: void AZ::Internal::EBusContainer<AZ::EntityEvents, AZ::EntityEvents, (AZ::EBusAddressPolicy)1, (AZ::EBusHandlerPolic
       │ y)1>::Dispatcher<AZ::EBus<AZ::EntityEvents, AZ::EntityEvents> >::Event<void (AZ::EntityEvents::*)(AZ::EntityId const&), AZ::
       │ EntityId&>(AZ::
 156   │ System: AzFramework::Application::StartCommon(AZ::Entity*) (+0x5a) [0x7fb179990aca]
 157   │ System: AzToolsFramework::ToolsApplication::StartCommon(AZ::Entity*) (+0x19) [0x7fb177dd48c9]
 158   │ System: EditorInternal::EditorToolsApplication::StartCommon(AZ::Entity*) (+0x1c) [0x7fb1765a619c]
 159   │ System: AzFramework::Application::Start(AZ::ComponentApplication::Descriptor const&, AZ::ComponentApplication::StartupParame
       │ ters const&) (+0x3c) [0x7fb179990a4c]
 160   │ System: AzToolsFramework::ToolsApplication::Start(AZ::ComponentApplication::Descriptor const&, AZ::ComponentApplication::Sta
       │ rtupParameters const&) (+0x4b) [0x7fb177dd482b]
 161   │ System: EditorInternal::EditorToolsApplication::Start() (+0x51) [0x7fb1765a6231]
 162   │ System: CryEditMain (+0x494) [0x7fb176663344]
 163   │ System: main (+0xe0) [0x564ae925f530]
 164   │ System: __libc_init_first (+0x90) [0x7fb17baa8d90]
 165   │ System: __libc_start_main (+0x80) [0x7fb17baa8e40]
 166   │ System: _start (+0x25) [0x564ae925f385]
 167   │ System: ==================================================================
 168   │ System: ====Assert added to ignore list by spec and verbosity setting.====
 169   │ RHISystem:
 170   │ ==================================================================
 171   │ RHISystem: Trace::Error
 172   │  /data/workspace/o3de/Gems/Atom/RHI/Code/Source/RHI/RHISystem.cpp(193): 'AZ::RHI::ResultCode AZ::RHI::RHISystem::InitInterna
       │ lDevices()'
 173   │ RHISystem: Failed to initialize RHI device.
 174   │ RHISystem: ===========
 175   │
 176   │ Segmentation fault (core dumped)
 ```

adamsj-ros, 2023-08-03:
Could this be from incompatible hardware? With an RTK model of Nvidia card but it will be using Docker in WSL2 which has not always been straightforward.

adamsj-ros, 2023-08-03:
Details on the old GeForce GTX 1050 that this error occured on vs the RTX A100 I will test on next. 

https://askgeek.io/en/gpus/vs/NVIDIA_A100-SXM4-40-GB-vs-NVIDIA_GeForce-GTX-1050-Max-Q

adamdbrw, 2023-08-04:
@spham-amzn could you take a look at this?

adamdbrw, 2023-08-07:
The Editor is not supported on WSL2 due to lack of Vulkan support: https://github.com/microsoft/WSL/issues/5636

adamsj-ros, 2023-08-07:
The same thing happens on an older PC I have with an NVIDIA GeForce GTX 1050. 

[This page](https://developer.nvidia.com/vulkan-driver) doesn't explicitly mention the Nvidia RTX A1000 amongst the other NVIDIA RTX AXXXX cards. It looks like the support for Vulkan is very limited. Is there a way to run without using Vulkan? If so, how would that change the build instructions.

adamsj-ros, 2023-08-07:
Also could you offer a few GPUs that you have used?

lumberyard-employee-dm, 2023-08-07:
Well the Minimum specs for a Nvidia Graphics Card is a GTX 1060 (https://www.docs.o3de.org/docs/welcome-guide/requirements/#minimum-hardware-specification), which maybe because of a a particular supported feature set in the drivers.
Someone from @o3de/sig-graphics-audio should have more insight into the GPU specs though and minimum driver level

spham-amzn, 2023-08-08:
> My mistake. This was on an older PC I have with an NVIDIA GeForce GTX 1050 running PopOS 22.04.
> 
> My Windows laptop running WSL2 has a Nvidia RTX A1000. [This page](https://developer.nvidia.com/vulkan-driver) doesn't explicitly mention that care amongst the other NVIDIA RTX AXXXX cards but the laptop is newly purchased about 6 months ago. It looks like the support for Vulkan might be very limited. Is there a way to run without using Vulkan? If so, how would that change the build instructions.

I think the issue is a combination of getting vulkan support through WSL2 and the drivers. There may be some extra work involved in getting it to work, as far as I know. ( I did find this [page](https://ubuntu.com/tutorials/enabling-gpu-acceleration-on-ubuntu-on-wsl2-with-the-nvidia-cuda-platform#1-overview) on nvidia's site)

adamdbrw, 2023-08-08:
@adamsj-ros we are running O3DE with RTX 3080 or stronger.

adamdbrw, 2023-08-31:
Closing this as WSL is not supported as a platform, because of lack of Vulkan support.

closed 2023-08-31 by adamdbrw