o3de/ROSConDemo #263: Update dockerfile for development to 2310

adamdbrw, 2023-10-09:
Once the branch is there on the release

michalpelka, 2023-12-07:
Since we are close to release 2310.1, I will update it after release of this version.

jeremy-kubota, 2023-12-08:
I'd also like to suggest that we keep the `Dockerfile` to contain revisions (hash or tag) instead of branches. If you use main branches and there are breaking API changes (such as how it is as of today), then the build will fail. If you pick a tag or hash, then it's at least a known good version.

jeremy-kubota, 2023-12-08:
Might also consider just creating submodules of this ROSConDemo for o3de and o3de-extras or if you want to continue to have development branches of those sub repos then the `vcs` `repos` file works best because you can use versions of hash, tag or branch. Branch is good when you have to move all three repos along during development on a feature branch of the main repo.

michalpelka, 2024-01-11:
Closing the issue, since it was resolved with updates to pull HEAD of the main branch for o3de and o3de-extras.

closed 2024-01-11 by michalpelka