o3de/ROSConDemo #265: SceneImportSettings.h is missing in o3de/o3de main branch and 23.10.0 Release .deb installer causing ROSConDemo project build to fail

LB-ArturZieba, 2023-10-10:
**Describe the bug**
If ROSConDemo project is built on the 23.10.0 Release .deb installer (https://o3debinaries.org/download/linux.html), or using o3de/o3de main branch engine (as listed in https://github.com/o3de/ROSConDemo/blob/development/README.md#o3de), then the project build will fail due to `SceneAPI/SceneCore/Import/SceneImportSettings.h` file missing.

```
[1237/1398] Building CXX object External/ROS2-5d5a250d/Code/CMakeFiles/ROS2.Editor.Static.dir/profile/Unity/unity_3_cxx.cxx.o
FAILED: External/ROS2-5d5a250d/Code/CMakeFiles/ROS2.Editor.Static.dir/profile/Unity/unity_3_cxx.cxx.o 
/usr/bin/clang++-14 -DAZ_BUILD_CONFIGURATION_TYPE=\"profile\" -DAZ_ENABLE_DEBUG_TOOLS -DAZ_ENABLE_TRACING -DAZ_PHYSICS_DEBUG_ENABLED -DAZ_PROFILE_BUILD -DDEFAULT_LY_PYTHONHOME=\"/home/gpc-800/build/o3de/python/runtime/python-3.10.5-rev4-linux/python\" -DDEFAULT_RMW_IMPLEMENTATION=rmw_fastrtps_cpp -DEDITOR_COMMON_IMPORTS -DENABLE_NON_COMPILED_CGF -DGRAPH_CANVAS_ENABLE_DETAILED_PROFILING=0 -DIMGUI_API_IMPORT -DIMGUI_ENABLED -DIMGUI_INCLUDE_IMGUI_USER_H -DLINUX -DLINUX64 -DNDEBUG -DPAL_TRAIT_LINUX_WINDOW_MANAGER_XCB -DPHYSX_EDITOR -DPHYSX_ENABLE_MULTI_THREADING -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_SVG_LIB -DQT_WIDGETS_LIB -DQT_XML_LIB -DRCUTILS_ENABLE_FAULT_INJECTION -DROS2_EDITOR -D_FORTIFY_SOURCE=2 -D_HAS_EXCEPTIONS=0 -D_PROFILE -D__linux__ -DCMAKE_INTDIR=\"profile\" -I/home/gpc-800/O3DE/Projects/ROSConDemo/Project/build/linux/External/ROS2-5d5a250d/Code/ROS2.Editor.Static_autogen/include_profile -I/home/gpc-800/O3DE/o3de-extras/Gems/ROS2/Code/Source -I/home/gpc-800/O3DE/o3de-extras/Gems/ROS2/Code/Source/RobotImporter -I/home/gpc-800/O3DE/o3de-extras/Gems/ROS2/Code/Include -I/home/gpc-800/build/o3de/Code/Tools/AssetProcessor/AssetBuilderSDK/. -I/home/gpc-800/build/o3de/Code/Framework/AzFramework/. -I/home/gpc-800/build/o3de/Code/Framework/AzFramework/Platform/Linux -I/home/gpc-800/build/o3de/Code/Framework/AzFramework/Platform/Common/Xcb -I/home/gpc-800/build/o3de/Code/Framework/AzNetworking/. -I/home/gpc-800/build/o3de/Code/Framework/AzNetworking/Platform/Common -I/home/gpc-800/build/o3de/Code/Framework/AzNetworking/Platform/Linux -I/home/gpc-800/O3DE/Projects/ROSConDemo/Project/build/linux/o3de/Code/Framework/AzNetworking/Azcg/Generated/AzNetworking -I/home/gpc-800/build/o3de/Code/Framework/AzToolsFramework/. -I/home/gpc-800/build/o3de/Code/Framework/AzToolsFramework/AzToolsFramework -I/home/gpc-800/build/o3de/Code/Framework/AzToolsFramework/Platform/Linux -I/home/gpc-800/build/o3de/Code/Framework/AzQtComponents/. -I/home/gpc-800/build/o3de/Code/Framework/AzQtComponents/Platform/Linux -I/home/gpc-800/build/o3de/Code/Framework/AzCore/. -I/home/gpc-800/build/o3de/Code/Framework/AzCore/Platform/Linux -I/home/gpc-800/build/o3de/Code/Framework/AzCore/Platform/Common -I/home/gpc-800/build/o3de/Gems/AtomLyIntegration/CommonFeatures/Code/Include -I/home/gpc-800/build/o3de/Gems/Atom/Tools/AtomToolsFramework/Code/Include -I/home/gpc-800/build/o3de/Gems/Atom/Feature/Common/Code/Include -I/home/gpc-800/build/o3de/Gems/Atom/Feature/Common/Code/3rdParty/ACES -I/home/gpc-800/build/o3de/Gems/Atom/RPI/Code/Include -I/home/gpc-800/build/o3de/Gems/Atom/RPI/Code/External -I/home/gpc-800/build/o3de/Code/Framework/AtomCore/. -I/home/gpc-800/build/o3de/Gems/Atom/RHI/Code/Include -I/home/gpc-800/build/o3de/Gems/Atom/Utils/Code/Include -I/home/gpc-800/build/o3de/Gems/Profiler/Code/Include -I/home/gpc-800/build/o3de/Gems/ImGui/Code/../External/ImGui/v1.82 -I/home/gpc-800/build/o3de/Gems/ImGui/Code/Include -I/home/gpc-800/build/o3de/Gems/AtomLyIntegration/AtomImGuiTools/Code/Include -I/home/gpc-800/build/o3de/Gems/LmbrCentral/Code/include -I/home/gpc-800/build/o3de/Gems/AudioSystem/Code/Include/Engine -I/home/gpc-800/build/o3de/Gems/GradientSignal/Code/Include -I/home/gpc-800/build/o3de/Gems/SurfaceData/Code/Include -I/home/gpc-800/build/o3de/Gems/Atom/Feature/Common/Code/Source -I/home/gpc-800/build/o3de/Gems/Atom/Asset/ImageProcessingAtom/Code/Include -I/home/gpc-800/build/o3de/Gems/Atom/Bootstrap/Code/Include -I/home/gpc-800/build/o3de/Gems/Atom/Component/DebugCamera/Code/Include -I/home/gpc-800/build/o3de/Gems/GraphCanvas/Code/. -I/home/gpc-800/build/o3de/Gems/GraphCanvas/Code/Source -I/home/gpc-800/build/o3de/Gems/GraphCanvas/Code/StaticLib -I/home/gpc-800/build/o3de/Gems/GraphModel/Code/Include -I/home/gpc-800/build/o3de/Code/Tools/SceneAPI/SceneCore/../.. -I/home/gpc-800/build/o3de/Code/Tools/SceneAPI/SceneData/../.. -I/home/gpc-800/build/o3de/Code/Editor/Include -I/home/gpc-800/build/o3de/Code/Editor/. -I/home/gpc-800/build/o3de/Code/Editor/.. -I/home/gpc-800/build/o3de/Code/Editor/Plugins/EditorCommon/. -I/home/gpc-800/build/o3de/Code/Legacy/CryCommon/. -I/home/gpc-800/build/o3de/Code/Legacy/CryCommon/.. -I/home/gpc-800/build/o3de/Code/Legacy/CryCommon/Platform/Linux -I/home/gpc-800/build/o3de/Gems/PhysX/Code/Include -I/home/gpc-800/build/o3de/Gems/PhysX/Code/NumericalMethods/Include -I/home/gpc-800/build/o3de/Gems/PhysX/Code/. -I/home/gpc-800/build/o3de/Gems/PhysX/Code/Source -I/home/gpc-800/build/o3de/Gems/PhysX/Code/Source/Platform/Linux -I/home/gpc-800/build/o3de/Gems/StartingPointInput/Code/Include -isystem /home/gpc-800/.o3de/3rdParty/packages/RapidJSON-1.1.0-rev1-multiplatform/RapidJSON/include -isystem /home/gpc-800/.o3de/3rdParty/packages/RapidXML-1.13-rev1-multiplatform/RapidXML/include -isystem /home/gpc-800/.o3de/3rdParty/packages/lz4-1.9.4-rev2-linux/lz4/include -isystem /home/gpc-800/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include -isystem /home/gpc-800/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtCore -isystem /home/gpc-800/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/./mkspecs/linux-g++ -isystem /home/gpc-800/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtGui -isystem /home/gpc-800/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtGui/5.15.1 -isystem /home/gpc-800/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtGui/5.15.1/QtGui -isystem /home/gpc-800/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtCore/5.15.1 -isystem /home/gpc-800/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtCore/5.15.1/QtCore -isystem /home/gpc-800/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtWidgets -isystem /home/gpc-800/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtWidgets/5.15.1 -isystem /home/gpc-800/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtWidgets/5.15.1/QtWidgets -isystem /home/gpc-800/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtSvg -isystem /home/gpc-800/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtXml -isystem /home/gpc-800/.o3de/3rdParty/packages/Lua-5.4.4-rev1-linux/Lua/include -isystem /home/gpc-800/.o3de/3rdParty/packages/zlib-1.2.11-rev5-linux/zlib/include -isystem /home/gpc-800/.o3de/3rdParty/packages/cityhash-1.1-multiplatform/cityhash/src -isystem /home/gpc-800/.o3de/3rdParty/packages/xxhash-0.7.4-rev1-multiplatform/xxhash/include -isystem /home/gpc-800/.o3de/3rdParty/packages/sdformat-13.5.0-rev2-linux/sdformat/include -isystem /home/gpc-800/.o3de/3rdParty/packages/sdformat-13.5.0-rev2-linux/sdformat/include/gz/sdformat13 -isystem /home/gpc-800/.o3de/3rdParty/packages/sdformat-13.5.0-rev2-linux/sdformat/include/gz/utils2 -isystem /home/gpc-800/.o3de/3rdParty/packages/sdformat-13.5.0-rev2-linux/sdformat/include/gz/math7 -isystem /home/gpc-800/.o3de/3rdParty/packages/png-1.6.37-rev2-linux/png/include -isystem /home/gpc-800/.o3de/3rdParty/packages/tiff-4.2.0.15-rev3-linux/tiff/include -isystem /home/gpc-800/build/o3de/python/runtime/python-3.10.5-rev4-linux/python/include/python3.10 -isystem /home/gpc-800/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtNetwork -isystem /home/gpc-800/.o3de/3rdParty/packages/PhysX-4.1.2.29882248-rev6-linux/PhysX/shared/include -isystem /home/gpc-800/.o3de/3rdParty/packages/PhysX-4.1.2.29882248-rev6-linux/PhysX/shared/include/foundation -isystem /home/gpc-800/.o3de/3rdParty/packages/PhysX-4.1.2.29882248-rev6-linux/PhysX/shared/include/geometry -isystem /home/gpc-800/.o3de/3rdParty/packages/poly2tri-7f0487a-rev1-linux/poly2tri/poly2tri/include/poly2tri -isystem /home/gpc-800/.o3de/3rdParty/packages/v-hacd-2.3-1a49edf-rev1-linux/v-hacd/v-hacd/include -isystem /opt/ros/humble/include/rclcpp -isystem /opt/ros/humble/include/ament_index_cpp -isystem /opt/ros/humble/include/libstatistics_collector -isystem /opt/ros/humble/include/builtin_interfaces -isystem /opt/ros/hu [... cut ...]
[O3DEROSConDemoSetupTerminalLog.txt](https://github.com/o3de/ROSConDemo/files/12856196/O3DEROSConDemoSetupTerminalLog.txt)

**Found in Branch**
.deb 23.10.0 Release installer (https://o3debinaries.org/download/linux.html)
o3de/o3de main (0d490a2b4b5e04fd17edf4850c91f94361d4b836)
o3de/o3de-extras development (9a692fd9fd4982283f0480244a42a26bf2f5993f)
o3de/ROSConDemo development (cdf1fe61d01ef301cb242c31f52ec4f5ed8ce0ef)

**Desktop**
 - Device: PC
 - OS: Linux
 - Version: Ubuntu 22.04.3 LTS
 - CPU AMD Ryzen 5 3600X 6-Core Processor
 - GPU AMD Radeon RX 5700 XT
 - Memory 16GB

LB-ArturZieba, 2023-10-10:
After reverifying this using the Development ([fc7f51a](https://github.com/o3de/o3de/commit/fc7f51a5748612eb069ec4623908311e2af99915)) for the ROSConDemo project setup, the build fails with error that seems to be related to the SceneImportSettings.h:
```
gpc-800@GPC-800:~/O3DE/Projects/ROSConDemo/Project$ cmake --build build/linux --config profile --target ROSConDemo Editor AssetProcessor ROSConDemo.Assets
[11/159] Building CXX object External/ROS2-5d5a250d/Code/CMakeFiles/ROS2.Editor.Static.dir/profile/Unity/unity_3_cxx.cxx.o
FAILED: External/ROS2-5d5a250d/Code/CMakeFiles/ROS2.Editor.Static.dir/profile/Unity/unity_3_cxx.cxx.o 
/usr/bin/clang++-14 -DAZ_BUILD_CONFIGURATION_TYPE=\"profile\" -DAZ_ENABLE_DEBUG_TOOLS -DAZ_ENABLE_TRACING -DAZ_PHYSICS_DEBUG_ENABLED -DAZ_PROFILE_BUILD -DDEFAULT_LY_PYTHONHOME=\"/home/gpc-800/build/o3de/python/runtime/python-3.10.5-rev4-linux/python\" -DDEFAULT_RMW_IMPLEMENTATION=rmw_fastrtps_cpp -DEDITOR_COMMON_IMPORTS -DENABLE_NON_COMPILED_CGF -DGRAPH_CANVAS_ENABLE_DETAILED_PROFILING=0 -DIMGUI_API_IMPORT -DIMGUI_ENABLED -DIMGUI_INCLUDE_IMGUI_USER_H -DLINUX -DLINUX64 -DNDEBUG -DPAL_TRAIT_LINUX_WINDOW_MANAGER_XCB -DPHYSX_EDITOR -DPHYSX_ENABLE_MULTI_THREADING -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_SVG_LIB -DQT_WIDGETS_LIB -DQT_XML_LIB -DRCUTILS_ENABLE_FAULT_INJECTION -DROS2_EDITOR -D_FORTIFY_SOURCE=2 -D_HAS_EXCEPTIONS=0 -D_PROFILE -D__linux__ -DCMAKE_INTDIR=\"profile\" -I/home/gpc-800/O3DE/Projects/ROSConDemo/Project/build/linux/External/ROS2-5d5a250d/Code/ROS2.Editor.Static_autogen/include_profile -I/home/gpc-800/O3DE/o3de-extras/Gems/ROS2/Code/Source -I/home/gpc-800/O3DE/o3de-extras/Gems/ROS2/Code/Source/RobotImporter -I/home/gpc-800/O3DE/o3de-extras/Gems/ROS2/Code/Include -I/home/gpc-800/build/o3de/Code/Tools/AssetProcessor/AssetBuilderSDK/. -I/home/gpc-800/build/o3de/Code/Framework/AzFramework/. -I/home/gpc-800/build/o3de/Code/Framework/AzFramework/Platform/Linux -I/home/gpc-800/build/o3de/Code/Framework/AzNetworking/. -I/home/gpc-800/build/o3de/Code/Framework/AzNetworking/Platform/Common -I/home/gpc-800/build/o3de/Code/Framework/AzNetworking/Platform/Linux -I/home/gpc-800/O3DE/Projects/ROSConDemo/Project/build/linux/o3de/Code/Framework/AzNetworking/Azcg/Generated/AzNetworking -I/home/gpc-800/build/o3de/Code/Framework/AzToolsFramework/. -I/home/gpc-800/build/o3de/Code/Framework/AzToolsFramework/AzToolsFramework -I/home/gpc-800/build/o3de/Code/Framework/AzToolsFramework/Platform/Linux -I/home/gpc-800/build/o3de/Code/Framework/AzFramework/Platform/Common/Xcb -I/home/gpc-800/build/o3de/Code/Framework/AzQtComponents/. -I/home/gpc-800/build/o3de/Code/Framework/AzQtComponents/Platform/Linux -I/home/gpc-800/build/o3de/Code/Framework/AzCore/. -I/home/gpc-800/build/o3de/Code/Framework/AzCore/Platform/Linux -I/home/gpc-800/build/o3de/Code/Framework/AzCore/Platform/Common -I/home/gpc-800/build/o3de/Gems/AtomLyIntegration/CommonFeatures/Code/Include -I/home/gpc-800/build/o3de/Gems/Atom/Tools/AtomToolsFramework/Code/Include -I/home/gpc-800/build/o3de/Gems/Atom/Feature/Common/Code/Include -I/home/gpc-800/build/o3de/Gems/Atom/Feature/Common/Code/3rdParty/ACES -I/home/gpc-800/build/o3de/Gems/Atom/RPI/Code/Include -I/home/gpc-800/build/o3de/Gems/Atom/RPI/Code/External -I/home/gpc-800/build/o3de/Code/Framework/AtomCore/. -I/home/gpc-800/build/o3de/Gems/Atom/RHI/Code/Include -I/home/gpc-800/build/o3de/Gems/Atom/Utils/Code/Include -I/home/gpc-800/build/o3de/Gems/Profiler/Code/Include -I/home/gpc-800/build/o3de/Gems/ImGui/Code/../External/ImGui/v1.82 -I/home/gpc-800/build/o3de/Gems/ImGui/Code/Include -I/home/gpc-800/build/o3de/Gems/AtomLyIntegration/AtomImGuiTools/Code/Include -I/home/gpc-800/build/o3de/Gems/LmbrCentral/Code/include -I/home/gpc-800/build/o3de/Gems/AudioSystem/Code/Include/Engine -I/home/gpc-800/build/o3de/Gems/GradientSignal/Code/Include -I/home/gpc-800/build/o3de/Gems/SurfaceData/Code/Include -I/home/gpc-800/build/o3de/Gems/Atom/Feature/Common/Code/Source -I/home/gpc-800/build/o3de/Gems/Atom/Asset/ImageProcessingAtom/Code/Include -I/home/gpc-800/build/o3de/Gems/Atom/Bootstrap/Code/Include -I/home/gpc-800/build/o3de/Gems/Atom/Component/DebugCamera/Code/Include -I/home/gpc-800/build/o3de/Gems/GraphCanvas/Code/. -I/home/gpc-800/build/o3de/Gems/GraphCanvas/Code/Source -I/home/gpc-800/build/o3de/Gems/GraphCanvas/Code/StaticLib -I/home/gpc-800/build/o3de/Gems/GraphModel/Code/Include -I/home/gpc-800/build/o3de/Code/Tools/SceneAPI/SceneCore/../.. -I/home/gpc-800/build/o3de/Code/Tools/SceneAPI/SceneData/../.. -I/home/gpc-800/build/o3de/Code/Editor/Include -I/home/gpc-800/build/o3de/Code/Editor/. -I/home/gpc-800/build/o3de/Code/Editor/.. -I/home/gpc-800/build/o3de/Code/Editor/Plugins/EditorCommon/. -I/home/gpc-800/build/o3de/Code/Legacy/CryCommon/. -I/home/gpc-800/build/o3de/Code/Legacy/CryCommon/.. -I/home/gpc-800/build/o3de/Code/Legacy/CryCommon/Platform/Linux -I/home/gpc-800/build/o3de/Gems/PhysX/Code/Include -I/home/gpc-800/build/o3de/Gems/PhysX/Code/NumericalMethods/Include -I/home/gpc-800/build/o3de/Gems/PhysX/Code/. -I/home/gpc-800/build/o3de/Gems/PhysX/Code/Source -I/home/gpc-800/build/o3de/Gems/PhysX/Code/Source/Platform/Linux -I/home/gpc-800/build/o3de/Gems/StartingPointInput/Code/Include -isystem /home/gpc-800/.o3de/3rdParty/packages/RapidJSON-1.1.0-rev1-multiplatform/RapidJSON/include -isystem /home/gpc-800/.o3de/3rdParty/packages/RapidXML-1.13-rev1-multiplatform/RapidXML/include -isystem /home/gpc-800/.o3de/3rdParty/packages/lz4-1.9.4-rev2-linux/l [... cut ...]
In file included from /home/gpc-800/O3DE/Projects/ROSConDemo/Project/build/linux/External/ROS2-5d5a250d/Code/CMakeFiles/ROS2.Editor.Static.dir/Unity/unity_3_cxx.cxx:15:
/home/gpc-800/O3DE/o3de-extras/Gems/ROS2/Code/Source/RobotImporter/Utils/SourceAssetsStorage.cpp:548:49: error: use of undeclared identifier 'importSettings'
        sceneDataImportGroup->SetImportSettings(importSettings);
                                                ^
1 error generated.
[15/159] Building CXX object External/ScriptCanvas-6bd70b39/Code/CMakeFiles/ScriptCanvas.Static.dir/profile/Unity/unity_14_cxx.cxx.o
ninja: build stopped: subcommand failed.
```

jhanca-robotecai, 2023-10-13:
The problem is generated by the misalignment in your configuration. File `#include <SceneAPI/SceneCore/Import/SceneImportSettings.h>` was added in _development_ of  _o3de_ and did not make it to the last release. It is already used by the code in the _development_ branch of _o3de-extras_. It should built correctly with the _main_ branches selected for both.

jhanca-robotecai, 2023-10-17:
I am happy you successfully built and launched the demo. Thanks for checking!

I believe the issue can be closed now, as it describes the error related to `SceneImportSettings.h`, which is explained by the difference in your source code branches. We might reopen it or open another one if needed.

LB-ArturZieba, 2023-10-18:
Closing the issue as per the above comment.

closed 2023-10-18 by LB-ArturZieba