o3de/ROSConDemo #269: RHISystem: Failed to initialize RHI device

LuisFGutierrez01, 2023-12-05:
I have been been able to run O3DE successfully using the dockerfile build provided in the git repository. However, now when I try it again, it fails giving me  MESA, Vulkan, libGL, and RHI errors when I try to run the O3DE editor. I am assuming there are some compatibility issues with firmware since I last used it. Any help would be appreciated.

Error terminal output:
```
root@a1caf06b1114:/data/workspace# /data/workspace/ROSConDemo/Project/build/linux/bin/profile/Editor
Module: Attempting to load module:/data/workspace/ROSConDemo/Project/build/linux/bin/profile/libEditorLib.so
Module: Success!
MESA: error: Failed to query drm device.
libGL error: glx: failed to create dri3 screen
libGL error: failed to load driver: iris
libGL error: failed to open /dev/dri/card0: No such file or directory
libGL error: failed to load driver: iris
Module: Attempting to load module:/data/workspace/ROSConDemo/Project/build/linux/bin/profile/libAtomViewportDisplayInfo.so
Module: Success!
Module: Attempting to load module:/data/workspace/ROSConDemo/Project/build/linux/bin/profile/libCamera.Editor.so
Module: Success!
Module: Attempting to load module:/data/workspace/ROSConDemo/Project/build/linux/bin/profile/libAtomLyIntegration_CommonFeatures.Editor.so
Module: Success!
Module: Attempting to load module:/data/workspace/ROSConDemo/Project/build/linux/bin/profile/libLmbrCentral.Editor.so
Module: Success!

...

Module: Success!
Module: Attempting to load module:/data/workspace/ROSConDemo/Project/build/linux/bin/profile/libSceneCore.so
Module: Success!
Module: Attempting to load module:/data/workspace/ROSConDemo/Project/build/linux/bin/profile/libSceneData.so
Module: Success!
Module: Attempting to load module:/data/workspace/ROSConDemo/Project/build/linux/bin/profile/libSceneBuilder.so
Module: Success!
Module: Attempting to load module:/data/workspace/ROSConDemo/Project/build/linux/bin/profile/libSceneCore.so
Module: Success!
System: Network layer initialized
System: AudioSystem created!
RHISystem: Initializing RHI...
RHISystem: 	Enumerated physical device: llvmpipe (LLVM 15.0.7, 256 bits)
RHISystem: 	Using physical device: llvmpipe (LLVM 15.0.7, 256 bits)
System: 
==================================================================
System: Trace::Assert
 /data/workspace/o3de/Gems/Atom/RHI/Vulkan/Code/Source/RHI/Vulkan.h(102): (139944332287232) 'void AZ::Vulkan::AssertSuccess(VkResult)'
System: ASSERT: Vulkan API method failed: Feature not present
System: ------------------------------------------------
System: AZ::Vulkan::Device::InitInternal(AZ::RHI::PhysicalDevice&) (+0xc1e) [0x7f460aa3232e]
System: AZ::RHI::Device::Init(int, AZ::RHI::PhysicalDevice&) (+0xb4) [0x7f45fa659e14]
System: AZ::RHI::RHISystem::InitInternalDevices(AZ::RHI::InitDevicesFlags) (+0x8b3) [0x7f45fa5ebb33]
System: AZ::RHI::RHISystem::InitDevices(AZ::RHI::InitDevicesFlags) (+0x45) [0x7f45fa5eb255]
System: AZ::RPI::RPISystem::Initialize(AZ::RPI::RPISystemDescriptor const&) (+0x9b) [0x7f45fa23c73b]
System: AZ::RPI::RPISystemComponent::Activate() (+0x254) [0x7f45f9f3b8c4]
System: AZ::ModuleManager::ActivateEntities(AZStd::vector<AZStd::shared_ptr<AZ::ModuleDataImpl>, AZStd::allocator> const&) (+0xb16) [0x7f474f1266d6]
System: non-virtual thunk to AZ::ModuleManager::OnEntityActivated(AZ::EntityId const&) (+0x94) [0x7f474f127064]
System: void AZ::Internal::EBusContainer<AZ::EntityEvents, AZ::EntityEvents, (AZ::EBusAddressPolicy)1, (AZ::EBusHandlerPolicy)1>::Dispatcher<AZ::EBus<AZ::EntityEvents, AZ::EntityEvents> >::Event<void (AZ::EntityEvents::*)(AZ::EntityId const&), AZ::EntityId&>(AZ:
System: AZ::Entity::Activate() (+0x12a) [0x7f474f668b4a]
System: AzFramework::Application::StartCommon(AZ::Entity*) (+0x5a) [0x7f475224e00a]
System: AzToolsFramework::ToolsApplication::StartCommon(AZ::Entity*) (+0x19) [0x7f475053d189]
System: EditorInternal::EditorToolsApplication::StartCommon(AZ::Entity*) (+0x1c) [0x7f474ec2dcdc]
System: AzFramework::Application::Start(AZ::ComponentApplication::Descriptor const&, AZ::ComponentApplication::StartupParameters const&) (+0x3c) [0x7f475224df8c]
System: AzToolsFramework::ToolsApplication::Start(AZ::ComponentApplication::Descriptor const&, AZ::ComponentApplication::StartupParameters const&) (+0x4b) [0x7f475053d0eb]
System: EditorInternal::EditorToolsApplication::Start() (+0x56) [0x7f474ec2dd76]
System: CryEditMain (+0x494) [0x7f474ecd0e64]
System: main (+0xe0) [0x559cf7a87680]
System: __libc_init_first (+0x90) [0x7f4754407d90]
System: __libc_start_main (+0x80) [0x7f4754407e40]
System: _start (+0x25) [0x559cf7a874d5]
System: ==================================================================
System: ====Assert added to ignore list by spec and verbosity setting.====
RHISystem: 
==================================================================
RHISystem: Trace::Error
 /data/workspace/o3de/Gems/Atom/RHI/Code/Source/RHI/RHISystem.cpp(202): 'AZ::RHI::ResultCode AZ::RHI::RHISystem::InitInternalDevices(AZ::RHI::InitDevicesFlags)'
RHISystem: Failed to initialize RHI device.
RHISystem: ==================================================================
Segmentation fault (core dumped)
```

Nvidia details:
```
❯ nvidia-smi
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 545.29.02              Driver Version: 545.29.02    CUDA Version: 12.3     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA RTX A1000 Laptop GPU    Off | 00000000:01:00.0 Off |                  N/A |
| N/A   53C    P8               4W /  40W |      8MiB /  4096MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A      4810      G   /usr/lib/xorg/Xorg                            4MiB |
+---------------------------------------------------------------------------------------+
```
Cuda compiler:
```
❯ nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2021 NVIDIA Corporation
Built on Thu_Nov_18_09:45:30_PST_2021
Cuda compilation tools, release 11.5, V11.5.119
Build cuda_11.5.r11.5/compiler.30672275_0
```
OS Version:
```
❯ cat /etc/os-release
NAME="Pop!_OS"
VERSION="22.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 22.04 LTS"
VERSION_ID="22.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=jammy
UBUNTU_CODENAME=jammy
LOGO=distributor-logo-pop-os
```

pijaro, 2023-12-06:
Could you show the output of `vulkaninfo`?

(If missing, install the vulkan utils with `apt install vulkan-utils`)

LuisFGutierrez01, 2023-12-06:
> Could you show the output of `vulkaninfo`?
> 
> (If missing, install the vulkan utils with `apt install vulkan-tools`)

Hello @pijaro, I have 3 detected devices showing up on vulkaninfo, I will only post the contents relating to the GPU I would like to use, the Nvidia RTX A1000:
```
❯ vulkaninfo                   
==========
VULKANINFO
==========

Vulkan Instance Version: 1.3.204

Instance Extensions: count = 20
===============================
	VK_EXT_acquire_drm_display             : extension revision 1
	VK_EXT_acquire_xlib_display            : extension revision 1
	VK_EXT_debug_report                    : extension revision 10
	VK_EXT_debug_utils                     : extension revision 2
	VK_EXT_direct_mode_display             : extension revision 1
	VK_EXT_display_surface_counter         : extension revision 1
	VK_EXT_swapchain_colorspace            : extension revision 4
	VK_KHR_device_group_creation           : extension revision 1
	VK_KHR_display                         : extension revision 23
	VK_KHR_external_fence_capabilities     : extension revision 1
	VK_KHR_external_memory_capabilities    : extension revision 1
	VK_KHR_external_semaphore_capabilities : extension revision 1
	VK_KHR_get_display_properties2         : extension revision 1
	VK_KHR_get_physical_device_properties2 : extension revision 2
	VK_KHR_get_surface_capabilities2       : extension revision 1
	VK_KHR_surface                         : extension revision 25
	VK_KHR_surface_protected_capabilities  : extension revision 1
	VK_KHR_wayland_surface                 : extension revision 6
	VK_KHR_xcb_surface                     : extension revision 6
	VK_KHR_xlib_surface                    : extension revision 6

Layers: count = 4
=================
VK_LAYER_INTEL_nullhw (INTEL NULL HW) Vulkan version 1.1.73, layer version 1:
	Layer Extensions: count = 0
	Devices: count = 3
		GPU id = 0 (Intel(R) Graphics (ADL GT2))
		Layer-Device Extensions: count = 0

		GPU id = 1 (llvmpipe (LLVM 15.0.7, 256 bits))
		Layer-Device Extensions: count = 0

		GPU id = 2 (NVIDIA RTX A1000 Laptop GPU)
		Layer-Device Extensions: count = 0

VK_LAYER_MESA_device_select (Linux device selection layer) Vulkan version 1.3.211, layer version 1:
	Layer Extensions: count = 0
	Devices: count = 3
		GPU id = 0 (Intel(R) Graphics (ADL GT2))
		Layer-Device Extensions: count = 0

		GPU id = 1 (llvmpipe (LLVM 15.0.7, 256 bits))
		Layer-Device Extensions: count = 0

		GPU id = 2 (NVIDIA RTX A1000 Laptop GPU)
		Layer-Device Extensions: count = 0

VK_LAYER_MESA_overlay (Mesa Overlay layer) Vulkan version 1.3.211, layer version 1:
	Layer Extensions: count = 0
	Devices: count = 3
		GPU id = 0 (Intel(R) Graphics (ADL GT2))
		Layer-Device Extensions: count = 0

		GPU id = 1 (llvmpipe (LLVM 15.0.7, 256 bits))
		Layer-Device Extensions: count = 0
[... 1769 lines cut ...]
------------------------------------------
	vulkanMemoryModel                             = true
	vulkanMemoryModelDeviceScope                  = true
	vulkanMemoryModelAvailabilityVisibilityChains = true

VkPhysicalDeviceWorkgroupMemoryExplicitLayoutFeaturesKHR:
---------------------------------------------------------
	workgroupMemoryExplicitLayout                  = true
	workgroupMemoryExplicitLayoutScalarBlockLayout = true
	workgroupMemoryExplicitLayout8BitAccess        = true
	workgroupMemoryExplicitLayout16BitAccess       = true

VkPhysicalDeviceYcbcr2Plane444FormatsFeaturesEXT:
-------------------------------------------------
	ycbcr2plane444Formats = true

VkPhysicalDeviceYcbcrImageArraysFeaturesEXT:
--------------------------------------------
	ycbcrImageArrays = true

VkPhysicalDeviceZeroInitializeWorkgroupMemoryFeatures:
------------------------------------------------------
	shaderZeroInitializeWorkgroupMemory = true

```

michalpelka, 2023-12-07:
GPU drivers are installed correctly and available. Since it is more of an issue with o3de, I asked on o3de's [discord ](https://discord.com/channels/805939474655346758/816043793576886273/1182341526400024627) for some help from @o3de/sig-graphics-audio .

adamdbrw, 2023-12-19:
@LuisFGutierrez01 cc @akioCL were you able to solve this issue?

LuisFGutierrez01, 2023-12-19:
Hello @adamdbrw ,

I was not able to get this working. 
I believe the issue lies somewhere within Rocker but I have stopped investigating.

I have also tried to get the ROSConDemo working by installing the demo. I followed the instructions on the [building this project](https://github.com/o3de/ROSConDemo#building-this-project) section of the repo and was also unsuccessful. However, I get a different issue, this time with Ninja: 
```
$ cmake --build build/linux --config profile --target ROSConDemo Editor AssetProcessor ROSConDemo.Assets
ninja: error: unknown target 'ROSConDemo'
```

I am currently trying to create the project manually by importing and building the project through the use of the [URDF files / guide](https://github.com/o3de/ROSConDemo/blob/development/docs/ImportingURDF/URDF_import_guide.md) included in the repo and any other documentation I can find.

Any advice would be appreciated.

jhanca-robotecai, 2023-12-19:
Hello @LuisFGutierrez01 
If you build the code directly on the machine (without using Docker), please try the updated instructions from https://github.com/o3de/ROSConDemo/pull/272. I tested it yesterday and it worked for me.

LuisFGutierrez01, 2023-12-27:
Hello @jhanca-robotecai 
I have tried out the newest version of the `Main` branch after you merged [#272](https://github.com/o3de/ROSConDemo/pull/272) and everything seems to work correctly now after doing a fresh `git clone`.

This was done with the versions of o3de based on your [comment in a different thread](https://github.com/o3de/ROSConDemo/issues/271#issuecomment-1851674841)
o3de: `point-release/23101` `dc6e2f7a23`
o3de-extras: `main f21f4812`

closed 2023-12-27 by LuisFGutierrez01