o3de/ROSConDemo #271: build failure

jeremy-kubota, 2023-12-11:
While building the latest on `development` using the [docker instructions](https://github.com/o3de/ROSConDemo/blob/development/docker/README.md) , I get the following error:

```
[1235/1399] Building CXX object External/ROS2-4e7b3125/Code/CMakeFiles/ROS2.Editor.Static.dir/profile/Unity/unity_3_cxx.cxx.o
FAILED: External/ROS2-4e7b3125/Code/CMakeFiles/ROS2.Editor.Static.dir/profile/Unity/unity_3_cxx.cxx.o 
/usr/bin/clang++-14 -DAZ_BUILD_CONFIGURATION_TYPE=\"profile\" -DAZ_ENABLE_DEBUG_TOOLS -DAZ_ENABLE_TRACING -DAZ_PHYSICS_DEBUG_ENABLED -DAZ_PROFILE_BUILD -DDEFAULT_LY_PYTHONHOME=\"/data/workspace/o3de/python/runtime/python-3.10.5-rev4-linux/python\" -DDEFAULT_RMW_IMPLEMENTATION=rmw_fastrtps_cpp -DEDITOR_COMMON_IMPORTS -DENABLE_NON_COMPILED_CGF -DGRAPH_CANVAS_ENABLE_DETAILED_PROFILING=0 -DIMGUI_API_IMPORT -DIMGUI_ENABLED -DIMGUI_INCLUDE_IMGUI_USER_H -DLINUX -DLINUX64 -DNDEBUG -DPAL_TRAIT_LINUX_WINDOW_MANAGER_XCB -DPHYSX_EDITOR -DPHYSX_ENABLE_MULTI_THREADING -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_SVG_LIB -DQT_WIDGETS_LIB -DQT_XML_LIB -DRCUTILS_ENABLE_FAULT_INJECTION -DROS2_EDITOR -D_FORTIFY_SOURCE=2 -D_HAS_EXCEPTIONS=0 -D_PROFILE -D__linux__ -DCMAKE_INTDIR=\"profile\" -I/data/workspace/ROSConDemo/Project/build/linux/External/ROS2-4e7b3125/Code/ROS2.Editor.Static_autogen/include_profile -I/data/workspace/o3de-extras/Gems/ROS2/Code/Source -I/data/workspace/o3de-extras/Gems/ROS2/Code/Source/RobotImporter -I/data/workspace/o3de-extras/Gems/ROS2/Code/Include -I/data/workspace/o3de/Code/Tools/AssetProcessor/AssetBuilderSDK/. -I/data/workspace/o3de/Code/Framework/AzFramework/. -I/data/workspace/o3de/Code/Framework/AzFramework/Platform/Linux -I/data/workspace/o3de/Code/Framework/AzFramework/Platform/Common/Xcb -I/data/workspace/o3de/Code/Framework/AzNetworking/. -I/data/workspace/o3de/Code/Framework/AzNetworking/Platform/Common -I/data/workspace/o3de/Code/Framework/AzNetworking/Platform/Linux -I/data/workspace/ROSConDemo/Project/build/linux/o3de/Code/Framework/AzNetworking/Azcg/Generated/AzNetworking -I/data/workspace/o3de/Code/Framework/AzToolsFramework/. -I/data/workspace/o3de/Code/Framework/AzToolsFramework/AzToolsFramework -I/data/workspace/o3de/Code/Framework/AzToolsFramework/Platform/Linux -I/data/workspace/o3de/Code/Framework/AzQtComponents/. -I/data/workspace/o3de/Code/Framework/AzQtComponents/Platform/Linux -I/data/workspace/o3de/Code/Framework/AzCore/. -I/data/workspace/o3de/Code/Framework/AzCore/Platform/Linux -I/data/workspace/o3de/Code/Framework/AzCore/Platform/Common -I/data/workspace/o3de/Gems/AtomLyIntegration/CommonFeatures/Code/Include -I/data/workspace/o3de/Gems/Atom/Tools/AtomToolsFramework/Code/Include -I/data/workspace/o3de/Gems/Atom/Feature/Common/Code/Include -I/data/workspace/o3de/Gems/Atom/Feature/Common/Code/3rdParty/ACES -I/data/workspace/o3de/Gems/Atom/RPI/Code/Include -I/data/workspace/o3de/Gems/Atom/RPI/Code/External -I/data/workspace/o3de/Code/Framework/AtomCore/. -I/data/workspace/o3de/Gems/Atom/RHI/Code/Include -I/data/workspace/o3de/Gems/Atom/Utils/Code/Include -I/data/workspace/o3de/Gems/Profiler/Code/Include -I/data/workspace/o3de/Gems/ImGui/Code/../External/ImGui/v1.82 -I/data/workspace/o3de/Gems/ImGui/Code/Include -I/data/workspace/o3de/Gems/AtomLyIntegration/AtomImGuiTools/Code/Include -I/data/workspace/o3de/Gems/LmbrCentral/Code/include -I/data/workspace/o3de/Gems/AudioSystem/Code/Include/Engine -I/data/workspace/o3de/Gems/GradientSignal/Code/Include -I/data/workspace/o3de/Gems/SurfaceData/Code/Include -I/data/workspace/o3de/Gems/Atom/Feature/Common/Code/Source -I/data/workspace/o3de/Gems/Atom/Asset/ImageProcessingAtom/Code/Include -I/data/workspace/o3de/Gems/Atom/Bootstrap/Code/Include -I/data/workspace/o3de/Gems/Atom/Component/DebugCamera/Code/Include -I/data/workspace/o3de/Gems/GraphCanvas/Code/. -I/data/workspace/o3de/Gems/GraphCanvas/Code/Source -I/data/workspace/o3de/Gems/GraphCanvas/Code/StaticLib -I/data/workspace/o3de/Gems/GraphModel/Code/Include -I/data/workspace/o3de/Code/Tools/SceneAPI/SceneCore/../.. -I/data/workspace/o3de/Code/Tools/SceneAPI/SceneData/../.. -I/data/workspace/o3de/Code/Editor/Include -I/data/workspace/o3de/Code/Editor/. -I/data/workspace/o3de/Code/Editor/.. -I/data/workspace/o3de/Code/Editor/Plugins/EditorCommon/. -I/data/workspace/o3de/Code/Legacy/CryCommon/. -I/data/workspace/o3de/Code/Legacy/CryCommon/.. -I/data/workspace/o3de/Code/Legacy/CryCommon/Platform/Linux -I/data/workspace/o3de/Gems/PhysX/Code/Include -I/data/workspace/o3de/Gems/PhysX/Code/NumericalMethods/Include -I/data/workspace/o3de/Gems/PhysX/Code/. -I/data/workspace/o3de/Gems/PhysX/Code/Source -I/data/workspace/o3de/Gems/PhysX/Code/Source/Platform/Linux -I/data/workspace/o3de/Gems/StartingPointInput/Code/Include -isystem /root/.o3de/3rdParty/packages/RapidJSON-1.1.0-rev1-multiplatform/RapidJSON/include -isystem /root/.o3de/3rdParty/packages/RapidXML-1.13-rev1-multiplatform/RapidXML/include -isystem /root/.o3de/3rdParty/packages/lz4-1.9.4-rev2-linux/lz4/include -isystem /root/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include -isystem /root/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtCore -isystem /root/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/./mkspecs/linux-g++ -isystem /root/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtGui -isystem /root/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtGui/5.15.1 -isystem /root/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtGui/5.15.1/QtGui -isystem /root/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtCore/5.15.1 -isystem /root/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtCore/5.15.1/QtCore -isystem /root/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtWidgets -isystem /root/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtWidgets/5.15.1 -isystem /root/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtWidgets/5.15.1/QtWidgets -isystem /root/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtSvg -isystem /root/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtXml -isystem /root/.o3de/3rdParty/packages/Lua-5.4.4-rev1-linux/Lua/include -isystem /root/.o3de/3rdParty/packages/zlib-1.2.11-rev5-linux/zlib/include -isystem /root/.o3de/3rdParty/packages/cityhash-1.1-multiplatform/cityhash/src -isystem /root/.o3de/3rdParty/packages/xxhash-0.7.4-rev1-multiplatform/xxhash/include -isystem /root/.o3de/3rdParty/packages/sdformat-13.5.0-rev2-linux/sdformat/include -isystem /root/.o3de/3rdParty/packages/sdformat-13.5.0-rev2-linux/sdformat/include/gz/sdformat13 -isystem /root/.o3de/3rdParty/packages/sdformat-13.5.0-rev2-linux/sdformat/include/gz/utils2 -isystem /root/.o3de/3rdParty/packages/sdformat-13.5.0-rev2-linux/sdformat/include/gz/math7 -isystem /root/.o3de/3rdParty/packages/png-1.6.37-rev2-linux/png/include -isystem /root/.o3de/3rdParty/packages/tiff-4.2.0.15-rev3-linux/tiff/include -isystem /data/workspace/o3de/python/runtime/python-3.10.5-rev4-linux/python/include/python3.10 -isystem /root/.o3de/3rdParty/packages/qt-5.15.2-rev9-linux/qt/include/QtNetwork -isystem /root/.o3de/3rdParty/packages/PhysX-5.1.1-rev1-linux/PhysX/physx/include -isystem /root/.o3de/3rdParty/packages/PhysX-5.1.1-rev1-linux/PhysX/physx/include/foundation -isystem /root/.o3de/3rdParty/packages/PhysX-5.1.1-rev1-linux/PhysX/physx/include/geometry -isystem /root/.o3de/3rdParty/packages/poly2tri-7f0487a-rev1-linux/poly2tri/poly2tri/include/poly2tri -isystem /root/.o3de/3rdParty/packages/v-hacd-2.3-1a49edf-rev1-linux/v-hacd/v-hacd/include -isystem /opt/ros/humble/include/rclcpp -isystem /opt/ros/humble/include/ament_index_cpp -isystem /opt/ros/humble/include/libstatistics_collector -isystem /opt/ros/humble/include/builtin_interfaces -isystem /opt/ros/humble/include/rosidl_runtime_c -isystem /opt/ros/humble/include/rcutils -isystem /opt/ros/humble/include/rosidl_typesupport_interface -isystem /opt/ros/humble/include/fastcdr -isystem /opt/ros/humble/include/rosidl_runtime_cpp -isystem /opt/ros/humble/include/rosidl_typesupport_fastrtps_cpp -isystem /opt/ros/humble/include/rmw -isystem /opt/ros/humble/include/rosidl_typesupport_fastrtps_c -isystem /opt/ros/humble/include/rosidl_typesupport_introspection_c -isystem /opt/ros/humble/include/rosidl_typesupport_introspection_cpp -isystem /opt/ros/humble/include/rcl -isystem /opt/ros/humble/include/rcl_interfaces -isystem /opt/ros/humble/include/rcl_logging_interface -isystem /opt/ros/humble/include/rcl_yaml_param_parser -isystem /op [... cut ...]
1 error generated.
```

I suspect that because the docker build is [using branches](https://github.com/o3de/ROSConDemo/blob/34051e129a05c7b2740017fd0513985099e57241/docker/Dockerfile#L18) and the versions are not locked, that an update to these branches has resulted in a build incompatibility. 

1. Which hashes/tags of `ROSConDemo`, `o3de`, and `o3de-extras` are compatible?
2. How do I determine compatible versions?
3. Can we lock to hashes in the `Dockerfile` instead of branches to avoid loss of compatibility?

floroeske, 2023-12-12:
Try reverting https://github.com/o3de/o3de-extras/commit/3b212e3dd96e05bd44ff8ca690966c46ebfae0ae in o3de-extras.

o3de(debian release "o3de_2310_0.deb" or main branch)
o3de-extras(main branch with 3b212e3 reverted) 
ROSConDemo(development branch)

jhanca-robotecai, 2023-12-12:
Sorry for the problem and confusion with the branches. We are waiting for the official release of 2310.1 before we clean this mess up (it should happen this Thursday). There were multiple breaking changes lately in o3de and it is difficult to keep it updated in all Gems/Demos due to our limited resources.

I recently tested the following configuration and it worked correctly:
o3de: `point-release/23101` dc6e2f7a23 (I believe the head of this branch should work as well)
o3de-extras: `main` f21f4812 (2310.1 was already released and it is available in `main`)
ROSConDemo: `development`

jhanca-robotecai, 2023-12-19:
@jeremy-kubota 
I tested the most recent code of #272 a8f7ced after the `2310.1` release and I managed to build all three docker images: `Dockerfile`, `Dockerfile.NavStack`, and `Dockerfile.Simulation`. I will close the issue now. Please reopen this issue if you find any issues.

closed 2023-12-19 by jhanca-robotecai

jeremy-kubota, 2023-12-19:
I think the real problem is that it may work one day and not the other because the default args are set to be branch heads instead of a fixed hash or tagged version. I still think there is a problem here if either the `o3de` or `o3de` main branches are updated in a way that's not compatible with the ROSConDemo. You need specific versions rather than branch HEADs for this to be a permanent fix.

jhanca-robotecai, 2023-12-19:
Thank you for the reply. I agree with your comment and with your solution. We use tags in the [other demo](https://github.com/RobotecAI/ROSCon2023Demo/blob/d7fe891ce577c706d4c1d94a90ffc53007dc30e0/README.md?plain=1#L90)

There is no agreement for this demo though. We will switch all branches to `main` (we will merge `development`->`main` after #272 is completed) and keep updating the code at every o3de release.