o3de/ROSConDemo #273: 'scripts/o3de.sh register -pp' not worked

servingrobotmaster, 2023-12-18:
robotmaster:~/o3de$ scripts/o3de.sh register -pp ~/ROSConDemo/Project/

[ERROR] o3de.manifest: Invalid engine json /snap/o3de/11/23.10.0/engine.json supplied or file missing.

I installed 23.10.1 with snapstore, but the error came up as follows, what should I do? As far as I know, there is no release of version 23.10.0.

jhanca-robotecai, 2023-12-18:
Please verify if paths in your o3de _manifest_ file are correct. Typically the _manifest_ file is stored under the name `~/.o3de/o3de_manifest.json`. 

In case of any troubles with this demo, I recommend using _Docker_ environment, please see the [READEM}(https://github.com/o3de/ROSConDemo/blob/development/docker/README.md)

servingrobotmaster, 2023-12-19:
Hello! Thank you for your response! It seems that my manifest.json path is correct. However, although there is a json for version 23.10.1, I'm curious why it's asking for the 23.10.0 json. Due to this issue, my Docker build also failed.

1428.9 -- Configuring incomplete, errors occurred!
1428.9 See also "/data/workspace/ROSConDemo/Project/build/linux/CMakeFiles/CMakeOutput.log".
------
Dockerfile:96
--------------------
  95 |     # Clone, register, and build the projects
  96 | >>> RUN . /opt/ros/${ROS_DISTRO}/setup.sh \
  97 | >>>     \
  98 | >>>     # Clone o3de \
  99 | >>>     \
 100 | >>>     && git clone --single-branch -b $O3DE_BRANCH $O3DE_REPO $O3DE_ROOT \
 101 | >>>     && git -C $O3DE_ROOT lfs install \
 102 | >>>     && git -C $O3DE_ROOT lfs pull \
 103 | >>>     && git -C $O3DE_ROOT reset --hard $O3DE_COMMIT \
 104 | >>>     \
 105 | >>>     ############################################################### \
 106 | >>>     # Apply a patch to fix a release compile issue if needed \
 107 | >>>     ############################################################### \
 108 | >>>     && if [ "$(md5sum $O3DE_ROOT/Gems/Atom/RPI/Code/Source/RPI.Public/RPISystem.cpp | awk '{print $1}')" = "07389c00378a13f31f1d6faf097a026b" ];then git -C $O3DE_ROOT apply $WORKSPACE/o3de_2.1.0.patch; fi \
 109 | >>>     \
 110 | >>>     ############################################################### \
 111 | >>>     # Clone o3de-extras   \
 112 | >>>     ############################################################### \
 113 | >>>     && git clone --single-branch -b $O3DE_EXTRAS_BRANCH $O3DE_EXTRAS_REPO $O3DE_EXTRAS_ROOT \
 114 | >>>     && git -C $O3DE_EXTRAS_ROOT lfs install \
 115 | >>>     && git -C $O3DE_EXTRAS_ROOT lfs pull \
 116 | >>>     && git -C $O3DE_EXTRAS_ROOT reset --hard $O3DE_EXTRAS_COMMIT \
 117 | >>>     \
 118 | >>>     ############################################################### \
 119 | >>>     # Clone the ROSConDemo \
 120 | >>>     ############################################################### \
 121 | >>>     && git clone --single-branch -b $ROSCON_DEMO_BRANCH $ROSCON_DEMO_REPO $ROSCON_DEMO_ROOT \
 122 | >>>     && git -C $ROSCON_DEMO_ROOT lfs install \
 123 | >>>     && git -C $ROSCON_DEMO_ROOT lfs pull \
 124 | >>>     && git -C $ROSCON_DEMO_ROOT reset --hard $ROSCON_DEMO_COMMIT \
 125 | >>>     \
 126 | >>>     ############################################################### \
 127 | >>>     # Get the O3DE python and register the gem paths and projects \
 128 | >>>     ############################################################### \
 129 | >>>     && $O3DE_ROOT/python/get_python.sh \
 130 | >>>     && $O3DE_ROOT/scripts/o3de.sh register -ep $O3DE_ROOT \
 131 | >>>     && $O3DE_ROOT/scripts/o3de.sh register -gp $O3DE_EXTRAS_ROOT/Gems/ROS2 \
 132 | >>>     && $O3DE_ROOT/scripts/o3de.sh register -pp $ROSCON_DEMO_PROJECT \
 133 | >>>     \
 134 | >>>     ############################################################### \
 135 | >>>     # Build the Editor, Tools, and clients
 136 | >>>     ############################################################### \
 137 | >>>     && cmake -B $ROSCON_DEMO_PROJECT/build/linux -G "Ninja Multi-Config" \
 138 | >>>                             -S $ROSCON_DEMO_PROJECT \
 139 | >>>                             -DLY_DISABLE_TEST_MODULES=ON \
 140 | >>>                             -DLY_STRIP_DEBUG_SYMBOLS=ON \
 141 | >>>                             -DAZ_USE_PHYSX5=ON \
 142 | >>>     && cmake --build $ROSCON_DEMO_PROJECT/build/linux --config profile \
 143 | >>>              --target ROSConDemo.GameLauncher Editor ROSConDemo.Assets \
 144 | >>>     \
 145 | >>>     ############################################################### \
 146 | >>>     # Build the Editor, Tools, and clients
 147 | >>>     ############################################################### \
 148 | >>>     && rm -rf ~/.o3de/3rdParty/ \
 149 | >>>     && rm -rf $O3DE_ROOT/.git \
 150 | >>>     && rm -rf $O3DE_ROOT/AutomatedTesting \
 151 | >>>     && rm -rf $O3DE_ROOT/python/downloaded_packages \
 152 | >>>     && rm -rf $ROSCON_DEMO_PROJECT/build/linux/Azcg/ \
 153 | >>>     && rm -rf $ROSCON_DEMO_PROJECT/build/linux/CMake \
 154 | >>>     && rm -rf $ROSCON_DEMO_PROJECT/build/linux/CMakeFiles/ \
 155 | >>>     && rm -rf $ROSCON_DEMO_PROJECT/build/linux/External/ \
 156 | >>>     && rm -rf $ROSCON_DEMO_PROJECT/build/linux/Testing/ \
 157 | >>>     && rm -rf $ROSCON_DEMO_PROJECT/build/linux/_deps/ \
 158 | >>>     && rm -rf $ROSCON_DEMO_PROJECT/build/linux/cmake \
 159 | >>>     && rm -rf $ROSCON_DEMO_PROJECT/build/linux/lib/ \
 160 | >>>     && rm -rf $ROSCON_DEMO_PROJECT/build/linux/o3de/ \
 161 | >>>     && rm -rf $ROSCON_DEMO_PROJECT/build/linux/packages/
 162 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c . /opt/ros/${ROS_DISTRO}/setup.sh             && git clone --single-branch -b $O3DE_BRANCH $O3DE_REPO $O3DE_ROOT     && git -C $O3DE_ROOT lfs install     && git -C $O3DE_ROOT lfs pull     && git -C $O3DE_ROOT reset --hard $O3DE_COMMIT         && if [ \"$(md5sum $O3DE_ROOT/Gems/Atom/RPI/Code/Source/RPI.Public/RPISystem.cpp | awk '{print $1}')\" = \"07389c00378a13f31f1d6faf097a026b\" ];then git -C $O3DE_ROOT apply $WORKSPACE/o3de_2.1.0.patch; fi         && git clone --single-branch -b $O3DE_EXTRAS_BRANCH $O3DE_EXTRAS_REPO $O3DE_EXTRAS_ROOT     && git -C $O3DE_EXTRAS_ROOT lfs install     && git -C $O3DE_EXTRAS_ROOT lfs pull     && git -C $O3DE_EXTRAS_ROOT reset --hard [... cut ...] ld/linux/CMake     && rm -rf $ROSCON_DEMO_PROJECT/build/linux/CMakeFiles/     && rm -rf $ROSCON_DEMO_PROJECT/build/linux/External/     && rm -rf $ROSCON_DEMO_PROJECT/build/linux/Testing/     && rm -rf $ROSCON_DEMO_PROJECT/build/linux/_deps/     && rm -rf $ROSCON_DEMO_PROJECT/build/linux/cmake     && rm -rf $ROSCON_DEMO_PROJECT/build/linux/lib/     && rm -rf $ROSCON_DEMO_PROJECT/build/linux/o3de/     && rm -rf $ROSCON_DEMO_PROJECT/build/linux/packages/" did not complete successfully: exit code: 1

servingrobotmaster, 2023-12-19:
Also, as stated on github: cmake -B build/linux -S . When I run -G "Ninja Multi-Config" -DLY_3RDPARTY_PATH=$HOME/o3de-packages, the following error appears and I am wondering how to resolve it.

CMake Error at cmake/FileUtil.cmake:139 (file):
  file failed to open for reading (No such file or directory):

    /home/kist/O3DE/Restricted/Engines/o3de/restricted.json
Call Stack (most recent call first):
  cmake/PAL.cmake:206 (ly_file_read)
  cmake/PAL.cmake:231 (o3de_find_restricted_folder)
  cmake/PAL.cmake:248 (o3de_restricted_path)
  CMakeLists.txt:37 (include)

servingrobotmaster, 2023-12-19:
hello. thank you for the reply.
After solving that part, I was able to perform the build. However, a problem occurred as follows.

When I entered build/linux/bin/profile/Editor in the terminal,

Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libEditorLib.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libAtomViewportDisplayInfo.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libCamera.Editor.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libAtomLyIntegration_CommonFeatures.Editor.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libLmbrCentral.Editor.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libGradientSignal.Editor.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libEditorModeFeedback.Editor.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libMaestro.Editor.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libPrefabBuilder.Builders.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libSceneProcessing.Editor.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libROS2.Editor.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libROSConDemo.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libAtomFont.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libAtomImGuiTools.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libAtom_Bootstrap.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libAtom_Component_DebugCamera.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libAtom_RHI_DX12.Private.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libAtom_RHI_DX12.Builders.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libAtom_RHI_Metal.Private.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libAtom_RHI_Metal.Builders.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libAtom_RHI_Vulkan.Private.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libAtom_RHI_Vulkan.Builders.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libAtom_Feature_Common.Editor.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libAtom_Feature_Common.Builders.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libAtom_RHI.Private.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libAtom_RHI_Null.Private.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libAtom_RHI_Null.Builders.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libAtom_RPI.Editor.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libAtom_RPI.Builders.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libEditorPythonBindings.Editor.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libGraphCanvas.Editor.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libGraphModel.Editor.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libImageProcessingAtom.Editor.so
[... 53 lines cut ...]
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libVegetation.Editor.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libSceneCore.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libSceneData.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libSceneBuilder.so
Module: Success!
Module: Attempting to load module:/home/kist/ROSConDemo/Project/build/linux/bin/profile/libSceneCore.so
Module: Success!
System: Network layer initialized
System: AudioSystem created!
AWSCoreConfiguration: 
==================================================================
AWSCoreConfiguration: Trace::Warning
 /home/kist/o3de/Gems/AWSCore/Code/Source/Configuration/AWSCoreConfiguration.cpp(114): 'void AWSCore::AWSCoreConfiguration::ParseSettingsRegistryValues()'
AWSCoreConfiguration: Failed to get AllowAWSMetadataCredentials setting, will default to false.
AWSCoreConfiguration: ==================================================================
Segmentation fault (core dumped)
The above results were obtained.

my environment:
roscondemo: development
o3de: main
o3de-extra: main

jhanca-robotecai, 2023-12-19:
Hello again and thank you for your interest in O3DE. Unfortunately, the information you provided is not enough to debug your problem. Moreover, I believe _Github issues_ channel of the _ROSCon demo_ is not the right place to solve individual problems with installing and running O3DE. Once again I encourage you to use _Docker_ environment in which everything is configured beforehand.

adamdbrw, 2023-12-19:
Lets try to find the issue, @servingrobotmaster. We need to start with understanding what it is that you are doing and what your platform is. Could you fill in these parts of an issue template? 

**Steps to reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '...'
3. Select attached asset '...'
4. Scroll down to '...'
5. See error

**Desktop/Device (please complete the following information):**
 - Device: [e.g. PC, Mac, iPhone, Samsung] 
 - OS: [e.g. Windows, macOS, iOS, Android, Linux, including distribution and version (e. g. Ubuntu 22.04 / Windows 11 22H2)]

 - Version [e.g. 10, Monterey, Oreo]
 - CPU [e.g. Intel I9-9900k , Ryzen 5900x, ]
 - GPU [AMD 6800 XT, NVidia RTX 3090]
 - Memory [e.g. 16GB]

**Additional context**
Add any other context about the problem here.

Do you have another version of O3DE installed as well, such as 2310? Could you attach your logs?

closed 2024-01-11 by michalpelka